mantidproject/mantid #18047: MantidPlotMatplotlibTest fails with python 3 and matplotlib 1.5

rosswhitfield, 2016-11-14:
The sporadic failing test on Ubuntu 16.04 when using python 3 is MantidPlotMatplotlibTest.py http://builds.mantidproject.org/job/master_clean-ubuntu-16.04-python3/

Ubuntu 14.04 doesn't see this problem because it's running matplotlib `v1.3.1`

I can reproduce this with `ctest --output-on-failure --repeat-until-fail 100 -R MantidPlotMatplotlibTest` and it will fail with
```
Test project /home/rwp/mantid/Release/py3
    Start 1563: MantidPlotMatplotlibTest.py
    Test #1563: MantidPlotMatplotlibTest.py ......***Failed    7.10 sec
FrameworkManager-[Notice] Welcome to Mantid 3.8.20161114.1104
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
QCoreApplication::exec: The event loop is already running
MantidPlot: Fatal IO error: client killed
```

It is failing on [`plt.close('all')`](https://github.com/mantidproject/mantid/blob/master/MantidPlot/test/MantidPlotMatplotlibTest.py#L26) and can be fixed by changing it to the same as [`gui_cmd(plt.close, 'all')`](https://github.com/mantidproject/mantid/blob/master/MantidPlot/test/MantidPlotMatplotlibTest.py#L31).

@martyngigg What is the best solution here? Can I just change it to use `gui_cmd(plt.close, 'all')` regardless of the matplotlib version?

martyngigg, 2016-11-15:
@rosswhitfield This indicates there is something wrong with the custom [matplotlib backend](https://github.com/mantidproject/mantid/blob/master/MantidPlot/pymantidplot/mpl/backend_mtdqt4agg.py) that I put in to be able to do away with that `gui_cmd` stuff. I'm glad I added this test now...

rosswhitfield, 2016-11-15:
That was the usual output but sometimes it fails via Timeout and sometimes fails while making a lot more noise:
```
*** Error in `/home/rwp/mantid/Release/py3/bin/MantidPlot': double free or corruption (fasttop): 0x00007fae0c20d080 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x70c4b)[0x7fae740aec4b]
/usr/lib/libc.so.6(+0x76fe6)[0x7fae740b4fe6]
/usr/lib/libc.so.6(+0x777de)[0x7fae740b57de]
/usr/lib/libX11.so.6(_XEventsQueued+0x55)[0x7fae6a6c1fe5]
/usr/lib/libX11.so.6(XFlush+0x1a)[0x7fae6a6a437a]
/usr/lib/libQtGui.so.4(_ZN14QWidgetPrivate8hide_sysEv+0x72)[0x7fae782e3da2]
/usr/lib/libQtGui.so.4(_ZN14QWidgetPrivate11hide_helperEv+0x94)[0x7fae7829ecb4]
/usr/lib/libQtGui.so.4(_ZN7QWidget10setVisibleEb+0x3d0)[0x7fae782a1d80]
/usr/lib/python3.5/site-packages/PyQt4/QtGui.so(+0x3b5e7c)[0x7fae59a3ee7c]
/usr/lib/libQtGui.so.4(_ZN14QWidgetPrivate12close_helperENS_9CloseModeE+0x1dc)[0x7fae7829f03c]
/usr/lib/python3.5/site-packages/PyQt4/QtGui.so(+0x5b2f25)[0x7fae59c3bf25]
/usr/lib/libpython3.5m.so.1.0(PyCFunction_Call+0xc9)[0x7fae76a688a9]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x86fb)[0x7fae76ae047b]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x5d8e)[0x7fae76addb0e]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x88ca)[0x7fae76ae064a]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x5d8e)[0x7fae76addb0e]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x88ca)[0x7fae76ae064a]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23)[0x7fae76ae13a3]
/usr/lib/libpython3.5m.so.1.0(+0x930f5)[0x7fae76a4c0f5]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0xee6)[0x7fae76ad8c66]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23)[0x7fae76ae13a3]
/usr/lib/libpython3.5m.so.1.0(+0x93008)[0x7fae76a4c008]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(+0x7ed44)[0x7fae76a37d44]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(+0xc50c0)[0x7fae76a7e0c0]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x3d6b)[0x7fae76adbaeb]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23)[0x7fae76ae13a3]
/usr/lib/libpython3.5m.so.1.0(+0x930f5)[0x7fae76a4c0f5]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0xee6)[0x7fae76ad8c66]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23)[0x7fae76ae13a3]
/usr/lib/libpython3.5m.so.1.0(+0x93008)[0x7fae76a4c008]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(+0x7ed44)[0x7fae76a37d44]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(+0xc50c0)[0x7fae76a7e0c0]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x3d6b)[0x7fae76adbaeb]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23)[0x7fae76ae13a3]
/usr/lib/libpython3.5m.so.1.0(+0x930f5)[0x7fae76a4c0f5]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0xee6)[0x7fae76ad8c66]
/usr/lib/libpython3.5m.so.1.0(+0x1282c3)[0x7fae76ae12c3]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalCodeEx+0x23)[0x7fae76ae13a3]
/usr/lib/libpython3.5m.so.1.0(+0x93008)[0x7fae76a4c008]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(+0x7ed44)[0x7fae76a37d44]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(+0xc50c0)[0x7fae76a7e0c0]
/usr/lib/libpython3.5m.so.1.0(PyObject_Call+0x47)[0x7fae76a212d7]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x3d6b)[0x7fae76adbaeb]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x88ca)[0x7fae76ae064a]
/usr/lib/libpython3.5m.so.1.0(PyEval_EvalFrameEx+0x88ca)[0x7fae76ae064a]
======= Memory map: ========
00400000-00d13000 r-xp 00000000 08:22 34472444                           /home/rwp/mantid/Release/py3/bin/MantidPlot
00f12000-00f13000 r--p 00912000 08:22 34472444                           /home/rwp/mantid/Release/py3/bin/MantidPlot
[... 324 lines cut ...]
MantidUI-[Notice] MantidPlot is shutting down...

Running tests...
----------------------------------------------------------------------
..
----------------------------------------------------------------------
Ran 2 tests in 0.268s

OK

Generating XML reports...
FrameworkManager-[Notice] Welcome to Mantid 3.8.20161115.1446
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
QCoreApplication::exec: The event loop is already running
QCoreApplication::exec: The event loop is already running
ApplicationWindow-[Notice] Script execution finished.
MantidU...
The rest of the test output was removed since it exceeds the threshold of 1024 bytes.
FrameworkManager-[Notice] Welcome to Mantid 3.8.20161115.1446
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
QCoreApplication::exec: The event loop is already running
MantidPlot: Fatal IO error: client killed

0% tests passed, 1 tests failed out of 1
```

martyngigg, 2016-11-15:
That definitely indicates an issue with the mechanism that ensures the Gui calls are on the main thread.  Does it happen with `python2` and `matplotlib` 1.5?

rosswhitfield, 2016-11-15:
No just python3, otherwise we would see this with out Ubuntu 16.04 pull requests.

martyngigg, 2016-11-17:
Interestingly I can't get this to fail with python 3.4 (I haven't updated my local machine to 16.04). This must be a `3.5` thing.

rosswhitfield, 2016-11-17:
Which version of matplotlib are you using? It only happens with [`>=1.5.0`](https://github.com/mantidproject/mantid/blob/master/MantidPlot/test/MantidPlotMatplotlibTest.py#L25). Ubuntu 14.04 is `1.3.1`.

martyngigg, 2016-11-17:
I installed `1.5.3` with `pip`

rosswhitfield, 2016-11-17:
OK then, must be a python 3.5 + matplotlib >= 1.5 problem.

rosswhitfield, 2016-11-18:
I just seen `MantidPlotMatplotlibTest` fail on Windows with what looks like the same problem? http://builds.mantidproject.org/job/pull_requests-win7/11210/console

martyngigg, 2016-12-02:
I can reproduce some strange behaviour with our custom backend in a standalone example (`main.cpp`):

```#include <Python.h>
#include <iostream>
#include <thread>
#include <fstream>

void run_pycode(const std::string & src) {
  PyRun_SimpleString(src.c_str());
}

void run_pycode_async(const std::string & src) {
  auto gil_s = PyGILState_Ensure();
  run_pycode(src);
  PyGILState_Release(gil_s);
}

int main(int argc, char**argv) {
  if (argc != 3) {
    std::cerr << "Usage:" << argv[0] << " python_file.py s|m\n";
    return 1;
  }

  const auto filename = argv[1];
  const auto mode = argv[2][0];
  std::ifstream fid(filename);
  std::string line, pycode;
  while(std::getline(fid, line)) {
    pycode += line + "\n";
  }

  Py_Initialize();
  // Start QApplication in main thread - simulates MantidPlot starting
  run_pycode("from PyQt4.QtGui import QApplication; app = QApplication(['main'])");
  run_pycode("import matplotlib as mpl; mpl.use(\"module://backend_mtdqt4agg\")");
  if(mode == 's') {
    run_pycode(pycode);
  } else {
    PyEval_InitThreads();
    auto main_ts = PyEval_SaveThread();
    std::thread worker(run_pycode_async, pycode);
    worker.join();
    PyEval_RestoreThread(main_ts);
  }
  Py_Finalize();
}```

Compile with (ubuntu 16.04):
```
g++ -std=c++14 -Wall -Wextra -I/usr/include/python3.5m main.cpp -lpthread -lpython3.5 -o main
```

Run it with:

```
PYTHONPATH=$PWD MPLBACKEND=qt4agg ./main MantidPlotMatplotlibTest.py s
```

`s` indicates single threaded, `m` multithreaded. It requires downloading [backend_mtdqt4agg.py](https://github.com/mantidproject/mantid/raw/master/MantidPlot/pymantidplot/mpl/backend_mtdqt4agg.py) to the same directory as `main`.

rosswhitfield, 2016-12-02:
I see that the single threaded works but multithreaded just hangs on the `worker.join()` line with no plots appearing. Is that what you are seeing?

martyngigg, 2016-12-05:
Actually yes it is so that's not the same thing. It looks like it's stuck with a call into Qt for some reason. I wonder if I shouldn't mix std::thread and with the Qt stuff.

martyngigg, 2017-01-18:
@rosswhitfield Can I push this back to `3.10`? I don't think I'm going to have much more time to look at this before Friday.

PR #18599 (2017-01-31): Fixes for custom matplotlib backend for v>=1.5
Description of work.

Updated matplotlib backend for v>=1.5 and hopefully fixed intermittent test failures. 

**To test:**

Find some [`matplotlib` examples](http://matplotlib.org/1.5.1/examples/index.html) and run them in the script window in MantidPlot. It should not crash.

Fixes #18047 

*Does not need to be in the release notes - internal fix*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quali [... cut]
closed 2017-01-31 by peterfpeterson