mantidproject/mantid #18094: Plotting: changing the Normalisation option to None doesn't rescale properly

DimitarTasev, 2016-11-22:
### Expected behavior
The whole plot is visible on both Normalisation options -> `None` and `Bin Width`

### Actual behavior
[image]

We **can't** fix what is shown by doing:
- Rescale to show all
- Zoom out

If we change the Normalisation back to `Bin Width` then the plot looks properly again.

### Steps to reproduce the behavior
1. Plot any workspace/spectra that has nice large peak (MAR11060 works)
2. Right click and got to Normalisation, and change that from `Bin Width` to `None`

### Platforms affected
Tested on Windows 7 and Ubuntu 14.04

DimitarTasev, 2016-11-22:
I'm leaving it open for anyone available to pick it up, as I am not sure when I will be able to

DTasev, 2019-04-08:
Probably not worth fixing now, with MantidPlot leaning towards deprecation and no reported issues for 2.5 years. Closing

closed 2019-04-08 by DTasev