mantidproject/mantid #18097: SplineInterpolation :: do linear interpolation when only two points are given

gvardany, 2016-11-23:
`SplineInterpolation` rightly throws an error when trying to interpolate 2 points with cubic spline. Generalisation would be relevant to do linear interpolation for the case of 2 points with a warning message instead of error.

reimundILL, 2016-11-24:
GSL can return the number of points required by the interpolation. That would be nice for validating the input of the algorithm (   unsigned int gsl_interp_type_min_size (const gsl_interp_type * T), unsigned int gsl_interp_min_size (const gsl_interp * interp)   )

reimundILL, 2016-11-29:
LinearBackground::functionDeriv1D seems never been used and looks incorrect. However, it was there and I think this method is good for completeness of the code.

PR #20028 (2017-08-31): SplineInterpolation to support linear
This PR extends the `SplineInterpolation` algorithm to support linear interpolation if only 2 points are present in the input. 

**To test:**
```
iws = CreateSampleWorkspace(NumBanks = 1, XMin = 7, XMax = 29, BinWidth = 11)
mws = CreateSampleWorkspace(NumBanks = 1, XMin = 6, XMax = 30, BinWidth = 3)

ows = SplineInterpolation(WorkspaceToMatch = mws, WorkspaceToInterpolate = iws, Linear2Points = True, DerivOrder = 0)
```

Then overlay the 1st spectrum of the `iws`, `mws` and `ows`.  Disable normalisation by bin width in the plot. Inside the range of `iws` t [... cut]
closed 2017-08-31 by martyngigg