mantidproject/mantid #18105: MultipleFileProperty in gui

peterfpeterson, 2016-11-23:
Reproduce the behaviour:
1. Set the logging level to `Information` so you can see `FileFinder` locate files
1. Use any algorithm that specifies a `MultipleFileProperty`. I've been using `SNSPowderReduction`, but any will do
1. Click on that property and type a list of runs that will force the system to use file finder. The more you specify, the more noticeable it will be. I've been using the string `NOM_81755-81770`.
1. Tab away to the `Browse` button and watch it not update the GUI. With this string it takes 3 seconds before I can interact with the GUI in any way

The best solution is if the use of `FileFinder` was in a separate thread from the GUI. I tracked it down to [`MultipleFileProperty::setValue`](https://github.com/mantidproject/mantid/blob/master/Framework/API/src/MultipleFileProperty.cpp#L115) calls [`MultipleFileProperty::setValueAsMultipleFiles`](https://github.com/mantidproject/mantid/blob/master/Framework/API/src/MultipleFileProperty.cpp#L217). Both of which are called in a single-threaded fashion. Unfortunately, `setValue` is not a unique enough name to track down where in the GUI code it is being called.

It was suggested that this was being updated through [`MWRunFiles`](https://github.com/mantidproject/mantid/blob/master/MantidQt/MantidWidgets/src/MWRunFiles.cpp), but that code appears to never be used anywhere. This issue is being used partially to help document where the call to `setValue` is being done so this performance issue can be fixed.

peterfpeterson, 2016-11-28:
It appears that `PropertyWidgetFactory` is returning the wrong type of object. It should return `MWRunFiles` rather than `FilePropertyWidget`. Unfortunately, because of linking `MWRunFiles` will need to be moved from `Widgets` to `API`. Also, `FilePropertyWidget` should be removed and replaced with `MWRunFiles`.

gvardany, 2019-09-27:
Was this not done a while ago? Shall we close this?

peterfpeterson, 2019-09-27:
There is a gui component for custom guis, but it isn't available by declaring a particular property type.

closed 2021-03-02 by stale[bot]