mantidproject/mantid #18151: Update fit benchmarking tables for release 3.9

Anders-Markvardsen, 2016-12-01:
Updating the tables in http://docs.mantidproject.org/nightly/concepts/FittingMinimizers.html to

* To include any new fitting test problems added and tested with the systemtest FittingBenchmarks
* Now we have tests against real neutron data problems - reorganised the concept page such tests with weights are at the forefront, the one users care most about, and non-weight test results can become a curiosity in the background
* Current the performance tables such that when Accuracy = NaN then also Performance = NaN

Anders-Markvardsen, 2017-01-20:
No further fit examples added to 3.9 hence move to 3.10

AnthonyLim23, 2017-11-02:
We do have two comparisons: http://docs.mantidproject.org/nightly/concepts/FittingMinimizers.html and http://docs.mantidproject.org/nightly/concepts/FittingMinimizersWeightedComparisonDetailed.html which were last updated for v3.8.0. I wonder if anybody knows where Fede's script for generating them went. I didn't find them in mantid's repo. merge into one page

Anders-Markvardsen, 2017-11-03:
The steps are to run:

1. https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/analysis/FittingBenchmarks.py but edited to run all the benchmarking tests

2. Manually copy and paste the new results tables into folder for that release

3. Manually update, 3 files I think it is, with the new release number

4. This is deliberately not automated because when updating the tables you may also need to update the top of http://docs.mantidproject.org/nightly/concepts/FittingMinimizers.html to say which minimizer is recommended are now recommended with the a given release - only if different of course. If not different this is not needed.

Writing this thinking... When we do change the recommended minimizer we should state from which release this applies

Anders-Markvardsen, 2018-05-16:
This issue was moved to mantidproject/fitbenchmarking#6

closed 2018-05-16 by Anders-Markvardsen