mantidproject/mantid #18159: Harmonize advanced plotting options

NickDraper, 2016-12-01:
### Current behavior

We have a few plotting options that are "advanced"
 1. waterfall plots - option on plot spectrum dialog
 2. tiled plots - option on plot spectrum dialog
 3. Plot Surface from group - option on workspace group context menu if 3 or more members 
 4. Plot Contour from group - option on workspace group context menu if 3 or more members
[image]

As you can see the approaches are a bit varied as to how you get to them (and the last two can be hard to find),but there is a lot of similarity in the dialog box they create.
[image]

[image]

### Actual behavior
We want to harmonize all of this without disrupting the well known and used plot Spectrum (+- errors) usage.

We want to replace the context menu options with 
 1. Plot Spectrum ...
 1. Plot Spectrum with errors ...
 1. Plot Special ...
 1. Color Fill plot (leave as before)

The two plot spectrum options had special behaviour if the workspace contained only one spectrum in that it would not show the dialog box and just plot immediately.  We want to retain that.

We want to harmonize the dialog box that the other options use, and plot special should be able to be used for waterfall and tiled plots as well as the surfaces and contours, and there is no reason it cannot also be used for normal plots.  Here is a suggested mockup.

[image]

The same dialog could be used for Plot Spectrum, just with the group box hidden and the form shrunk appropriately.

You might want to have a second level of context menus under plot special, they should just fill the plot type out in the dialog box for you.

All of the underlying plotting for this should already be in place.

### Platforms affected
All

KarlPalmen, 2016-12-08:
I'd appreciate an example file to load that produces `Plot ... from Group` in the right click menu. The MUSR files I tried, which have a group of 2 don't produce this is the right click menu.

KarlPalmen, 2016-12-08:
I find I can produce such a workspace by running the GroupWorkspace algorithm.

raquelalvarezbanos, 2016-12-08:
Hi @KarlPalmen, for real datasets try with `POLREF00003014`, it's a POLREF run with 32 periods which can be found in `\ExternalData\Testing\Data\SystemTest\`

KarlPalmen, 2016-12-08:
Think you @raquelalvarezbanos .

KarlPalmen, 2017-01-24:
Through further efforts, I have found that the disharmony between the two dialog boxes that need to be merged runs deep into the code and so will be postponing this issue into the following release.

NickDraper, 2017-02-17:
This script might be helpful to create temperature range data to test the Plotting against a workspace against a temperature range
``` python
base = CreateSampleWorkspace(Function='Powder Diffraction', NumBanks=1, BankPixelWidth=1)

wsNames = []
for t in range (10,110,10):
    wsName = "temp" + str(t)
    wsNames.append(wsName)
    MultiplyRange(base,75,99,t/100.0,OutputWorkspace=wsName)
    AddSampleLog(wsName,"Temp",str(t),"Number","Kelvin")
TestGroup = GroupWorkspaces(wsNames)
```

PR #19237 (2017-05-17): Harmonise plotting of workspaces from right-button click
Until this, surface and contour plotting was done in quite a different way to the plotting of spectra with or without errors in simple 1D, waterfall or tiled plot. Now they should all be accessible from a `Plot Spectrum Advanced` right-click menu option and basic plot spectrum functionality should be available as before.

The issue #18159 shows this in more detail.

**To test:**

The issue shows a python script and file to load that would be useful in testing also use your own file or scripts as you see fit.

Test the right click menu fo [... cut]
closed 2017-05-17 by quantumsteve