mantidproject/mantid #18172: Extend fit benchmarking script to be callable with any minimizers

Anders-Markvardsen, 2016-12-05:
Extend the fit benchmarking scripts introduced with #17654 so that the benchmarking scripts can be called with non-Mantid minimizers also.

There are several motivations for this:

* To make it easy to test if e.g. any of the minimizers in scipy or Matlab are better than the once we currently offer to our Mantid users 
* To enable someone to test new idea for a minimizer within needing to first learn how to add a minimizer in Mantid (Although IMPORTANTLY note testing against an entirely different code is not 100% the same as comparing minimizers within the mantid framework - although this - should! - not be that different - if there are differences this would be something Mantid could learn from also)
* Motivate more to use our benchmarking framework and extending it with more fit problems

AnthonyLim23, 2017-07-26:
this will need to use the functionality from #19010

AnthonyLim23, 2017-09-13:
https://stackoverflow.com/questions/10354163/assigning-a-function-to-a-variable should be useful to avoid assignment within the for loops

AnthonyLim23, 2017-10-18:
I have written a python script, with help from Roman. This demonstrates how we could do this. The plot is not done properly so may double up on data if a run fails. 

```
from scipy.optimize import curve_fit,leastsq
import matplotlib.pyplot as plt
import numpy as np
import mantid.simpleapi as msapi
from scipy.optimize import curve_fit,leastsq

# make the data
def func(x,coeffs):
    a = coeffs[0]
    b= coeffs[1]
    c= coeffs[2]
    return a*np.exp(-(x-b)*(x-b)/(c*c))

xData = np.linspace(0,4,50)
coeffs = [2.5,1.5,0.5]
y= func(xData,coeffs)
y_noise = 0.2*np.random.normal(size=xData.size)
yData = y + y_noise
sigma = np.sqrt(y_noise*y_noise)/2
ws = CreateWorkspace(xData,yData,sigma)
plt.errorbar(xData,yData,yerr=sigma,color='k',label='data')

# fitting stuff

# resets the function
def getFunction():
     return Gaussian(Height=1, Sigma=0.1,PeakCentre=0.5)

def pythonFit(fun,ws):
     print "python"
     #extract values from ws
     xData = ws.readX(0)
     yData = ws.readY(0)
     EData = ws.readE(0)
     # get the intial parameters
     p0=[]
     for val in fun:
      p0.append(val)
     #do the fit
     popt, pcov = curve_fit(fun, xData, yData, p0=p0, sigma=sigma)
     # get a mantid fit output 
     out =  msapi.Fit(fun, ws, Output='ws_fitting_test',
                           CostFunction="Least squares",
                           Minimizer="BFGS",MaxIterations=0,
                           IgnoreInvalidData=True,
                           StartX=xData[0],EndX=xData[-1])
     return out
     
def mantidFit(fun,ws):
    print "mantid"
    out =  msapi.Fit(fun, ws, Output='ws_fitting_test',
                           CostFunction="Least squares",
                           Minimizer="BFGS",MaxIterations=10000,
                           IgnoreInvalidData=True,
                           StartX=xData[0],EndX=xData[-1])
    # to be fair lets do zero iterations
    out =  msapi.Fit(out.Function, ws, Output='ws_fitting_test',
                           CostFunction="Least squares",
                           Minimizer="BFGS",MaxIterations=0,
                           IgnoreInvalidData=True,
                           StartX=xData[0],EndX=xData[-1])
    return out

# pick which fit to do
def getFit(min):
    if min =="python":
        return pythonFit
    elif min =="mantid":
        return mantidFit
    else:
        print "UNKOWN option"
        return

minimizers=["python","mantid"]
colour =["r","b"]
for k,min in enumerate(minimizers):
    doFit = getFit(min) 
    # could put a loop here over all of the data that uses min
    fun = getFunction()
    out = doFit(fun,ws)
    print min, out.Function,out.OutputChi2overDoF
    outWS= out.OutputWorkspace
    plt.plot(outWS.readX(1),outWS.readY(1),colour[k],label=min)
plt.legend()
plt.show()
```

Anders-Markvardsen, 2018-05-16:
This issue was moved to mantidproject/fitbenchmarking#5

closed 2018-05-16 by Anders-Markvardsen