mantidproject/mantid #18185: Fix some issues in Reflectometry GUI

raquelalvarezbanos, 2016-12-06:
This refers to the old reflectometry GUI. Issues were reported by Max:

- Processing creates a <runno>_IvsQ workspace which is re-binned according to the setting, but we'd like to have the un-binned IvsQ workspace too and the re-binned one should be called <runno>_IvsQ_binned, like it used to.
- "Transfer" button doesn't work.
- "Save Workspaces" dialog doesn't populate the path correctly, it's missing a slash, e.g.: U:Skoda

PranavBahuguna, 2016-12-08:
Regarding the first issue (retrieving unbinned ws), what should be done about grouping? Should the unbinned workspaces be grouped too? If not, should the group workspace for binned workspaces keep its name to "..._IvsQ", or should that change to "..._IvsQ_binned"?

raquelalvarezbanos, 2016-12-08:
I think the only possibility to get the unbinned output is to run `ConvertUnits` on the `IvsLam` workspace that `ReflectometryReductionOneAuto` outputs.

PranavBahuguna, 2016-12-08:
What should I do about the grouping issue?

raquelalvarezbanos, 2016-12-08:
Is the output `IvsLam` grouped?

PranavBahuguna, 2016-12-08:
Yes. My issue was whether or not we should group the original un-binned workspaces as well as the binned ones.

raquelalvarezbanos, 2016-12-08:
For the binned workspaces: I think that should be done automatically (`ReflectometryReductionOneAuto` will take care of that).

For the un-binned workspaces: I think if the `IvsLam` workspace is a workspace group, if we just run `ConvertUnits` on it a workspace group will be produced.

Let me know if I am wrong.

raquelalvarezbanos, 2017-02-06:
Not sure why this wasn't closed when https://github.com/mantidproject/mantid/pull/18239 was merged.

PR #18239 (2017-01-30): Fixed various issues in the old Reflectometry GUI
As stated in the relevant issue, this PR fixes the following issues in the old Reflectometry GUI:
- Processing creates a _IvsQ workspace which is re-binned according to the setting, but we'd like to have the un-binned IvsQ workspace too and the re-binned one should be called _IvsQ_binned, like it used to.
- "Transfer" button doesn't work.
- "Save Workspaces" dialog doesn't populate the path correctly, it's missing a slash, e.g.: U:Skoda

**To test:**
- The GUI can be accessed from Interfaces->Reflectometry->ISIS Reflectometry
- Load runs using  [... cut]
closed 2017-02-06 by raquelalvarezbanos