mantidproject/mantid #18226: Improve plots in Muon Analysis interface

raquelalvarezbanos, 2016-12-08:
This was requested by James:

> A minor style issue: a tiled window with 16 sub-plots wastes quite a lot of space for legends, which are wider than they need to be because of the verbose workspace names such as “MuonSimulFit_113041-4_HIFI00113041_long_1_Workspace”. In the case where all workspaces making up a tiled plot have the same set of spectrum names, one legend (on the first graph) with just “Data”,”Calc” and “Diff” will do, then add a label “113041 P1” to each subplot. If the rows/columns match up with runs/periods (as they do here) then you could just make the Y axis labels on the left say “Asymmetry run 113041” etc and the X axis labels such as “Time (us) period 3”.

> I haven’t yet investigated all the permutations but it would be nice if, when fitting N runs and P periods (or N runs and G groups, etc) the tiled plot arranges itself in a N*P or P*N grid with rows and columns lined up sensibly. 6 runs * 4 periods arranges the tiles in a 5*5 array with one blank space at the bottom right.

> (I can of course do Graph – Arrange Layers afterwards)

> The tiled graph starts with all scales equal. Should I be able to re-scale one of the graphs (perhaps the top left one) and have the rest follow it?

Anders-Markvardsen, 2017-01-09:
Not specific to Muon

Anders-Markvardsen, 2017-01-09:
Supply python script to take legends of is sufficient.

closed 2021-03-02 by stale[bot]