mantidproject/mantid #18228: ParameterMap: Extract and reinsert data for Instrument-2.0 transition

SimonHeybrock, 2016-12-08:
Instrument-2.0 will store certain information such as mask flags and position information in class like `SpectrumInfo` and `DetectorInfo`. Currently this data is stored in `ParameterMap` (only offsets for positions).

For compatibility with legacy files we need to be able to extract this data from the `ParameterMap`, reroute insertion of this data to new destinations, and reinsert this data into a `ParameterMap` before saving.

The following locations have been identified as relevant:

- Loading parameters from an IDF: This is done by several load algorithms. Uses `ExperimentInfo::populateInstrumentParameters()`, which in turn calls `ExperimentInfo::populateWithParameter()`. Redirecting information seems to be straight forward.

- `LoadNexusProcessed`: Uses `ExperimentInfo::readParameterMap()`. `Can filter and redirect data (`tokens[1]` is the parameter type, `tokens[2]` is the parameter name). Seems straight forward.

- `LoadMD`: Uses `MDBoxFlatTree::loadExperimentInfos`. This redirects either directly to `ExperimentInfo::readParameterMap()` (see above), or creates `FileBackedExperimentInfo`, which will eventually also call that method. Seems straight forward.

- `SaveNexusProcessed`: Uses `ExperimentInfo::saveExperimentInfoNexus()`, which calls `Instrument::saveNexus()`. We can solve this one by creating a copy of the instrument, insert data from `DetectorInfo` into the parameter map of the temporary instrument, and call `saveNexus()` on that. Seems straight forward.

@ianbush @OwenArnold: I had a look at the mechanism we discussed, it seems to be surprisingly easy, basically all changes are within `ExperimentInfo`. Can you think of any other cases? Did I overlook anything?

OwenArnold, 2016-12-08:
@SimonHeybrock. No nothing else comes to mind.

SimonHeybrock, 2016-12-09:
Before we do this we need to address the following:

- Code using `ExperimentInfo::replaceInstrumentParameters()` and `ExperimentInfo::swapInstrumentParameters()` needs to be able to also replace data that is then in `DetectorInfo`. `CopyInstrumentParameters` is the only client, so this is easy.

- `ExperimentInfo::getInstrument()` and `ExperimentInfo::setInstrument()` needs to somehow also track data in `DetectorInfo`.

PR #18318 (2017-01-04): Store masking information in class DetectorInfo
This PR moves the storage location of mask flags for detectors from `ParameterMap` into `Beamline::DetectorInfo`. This is part of Instrument-2.0.

Masking data is intercepted when loading and added to `DetectorInfo` instead of `ParameterMap`. Masking data is injected into `ParameterMap` before saving. This ensures **backward and forward compatibility** for files (in particular processed Nexus).

For compatibility with legacy Python scripts a wrapper for `IDetector::isMasked` is provided. As a consequence docstests that are using this method are u [... cut]
closed 2017-01-04 by peterfpeterson