mantidproject/mantid #18246: Follow up tasks for SANS reduction backend development

AntonPiccardoSelg, 2016-12-12:
As a follow up to the new ISIS SANS design and implementation, Richard and Rob have had the following suggestiosn which are partially already captured in other tickets

The rewrite should also address:

* inverse masking. Often a region of interest is very small (some slice settings for example). In these cases we should investigate if inverse masking could improve performance

* have wavelength scanning reductions

* get user file as a python version

* time masking on a detector-by-detector basis

For more details see the discussion below and ask Rob and Richard:

> I have not previously seen the specification document for the backend rebuild, Anton has obviously put in a lot of work here! Have the SANS team gone through all the flow diagrams carefully?

https://github.com/mantidproject/documents/blob/ec61dfdd306524022cc14dcfa7e4df6351967e7c/Design/SANS/NewISISDesign/Design_Document.md

Some thoughts that might make things go faster, but might also be tricky or impossible to implement ..

The “move instrument to correct position” is often the same for many runs in turn, with appropriate dependencies and state monitoring it may be possible to recognise this.

> I'm not sure the moving of things is a bottle neck really. I'd be more worried about the detector tube calibration file which is certainly always the same between calibrations.
> It might be a better idea in the case of the detector calibration to make sure the detector tables are corrected before even making it to mantid. This is possible but we haven't done it. You could then optionally apply a calibration if needed.

Likewise masking may also be the same for many runs in turn.

Then there is still the thought that, especially for transmissions on the main detector, inverse masking, ie saying what you want to include, rather than exclude, would be faster. However a good implementation of the previous point would likely mitigate this.

> Inverse masking would certainly be good for the stuff we were doing with the steels where narrow phi sectors are used +/- 10deg

A “loop over wavelength ranges” w0, w1, w2, w3, … wn can perhaps be done faster on the fly inside the main reduction for w0 to wn. If it could be done efficiently, then we could do this for all reductions and  automatically flag up poor wavelength overlap as an indicator of something being wrong or strong  multiple scatter. [I expect there is a ticket for this somewhere…]

> Again this would be very useful for quite a few thing that are becoming more common. e.g. metals where multiple scattering from Bragg is a big problem and as we try to push to measure SANS and SESANS samples which tend to scatter v strongly. The more interesting part of this would be to use the information to somehow correct for the MS in a model. This would require MC though. It is done in wide angle diffraction.

Other topics …

Are we going to change the user file to say look like python in parallel with the new back end, or would that be best left as a separate issue?

> I'd forgotten this, it would certainly be a good idea to make sure it is in the workflow structures.

There may be demand for masking time channels only in certain regions of the detector.

> This is arguably essential for work with cryostats with sapphire windows. Mostly at the moment I've been masking a small spot on the detector but there really is no need as the peaks are very narrow in time

smk78, 2016-12-12:
There are some other corrections that we do not currently apply to ISIS SANS reductions but which (variously) SNS SANS, HFIR SANS and EQSANS do:

https://github.com/mantidproject/mantid/issues/9937 (detector deadtime corrections)
https://github.com/mantidproject/mantid/issues/12087 (detector dark current); also see #16342
https://github.com/mantidproject/mantid/issues/12091 (sample self absorbtion)

It would be good to incorporate these as part of the reduction re-write.

smk78, 2020-07-16:
Of the tasks mentioned in this ticket:

- inverse masking: can now be done
- wavelength scanning reductions: is covered by https://github.com/mantidproject/mantid/issues/28987
- python user file: superseded by decision to use TOML
- detector-by-detector time masking: is covered by https://github.com/mantidproject/mantid/issues/10676
- detector deadtime corrections: is covered by https://github.com/mantidproject/mantid/issues/9937
- detector dark current corrections: is covered by https://github.com/mantidproject/mantid/issues/12087
- sample self-absorbtion corrections: is covered by https://github.com/mantidproject/mantid/issues/12087

So am closing this issue.

closed 2020-07-16 by smk78