mantidproject/mantid #18256: Waterfall & Tiled plots to be selected by drop-down list

KarlPalmen, 2016-12-12:
Currently Waterfall and Tiled plots are selected by separate check boxes. It does make sense to select both and therefore it makes sense to instead select between three (1D plot, Waterfall plot, Tiled plot).

This issue is part of #18159, within which additional options will later be added to this drop-down list.

KarlPalmen, 2016-12-21:
While testing this, I found out that if one is plotting several spectra from a group workspace, a tiled plot will have one tile per group member each with several spectra plotted. In such a circumstance the user might want to plot these as waterfall plots as is possible at present.

martyngigg, 2016-12-21:
@KarlPalmen Can we have another option in the list that is only present in this type of circumstance named something like "Tiled Waterfall"?

KarlPalmen, 2016-12-21:
I'll look into this "Tiled Waterfall" for group workspace possibility.

KarlPalmen, 2016-12-21:
I've found out from recent test build of Mantid that if you set both the Waterfall and Tiled checkboxes in such a situation, the tiled plot does not have any waterfall plotting done. Therefore this issue can proceed and Martyn's suggestion can be carried out as part of any work to make tiled waterfall plots available for workspace groups.

PR #18342 (2017-01-11): New widget for Waterfall & Tiled plots
When plotting spectra, change selection of Waterfall and Tiled plots from two check boxes to one dropdown list (ComboBox). Note that previous situation if both tiled and waterfall were selected,  tiled is plotted without waterfall.

**To test:**

Load or create various workspaces with more than one spectrum including group workspaces, 

check the drop-down list of 1D-plot, Waterfall Plot & Tiled Plot is available if you plot spectra or plot spectra with errors.

Check that it works as expected

You may also check that the dialog box does not appear, if on [... cut]
closed 2017-01-11 by martyngigg