mantidproject/mantid #18291: Performance regression related to #17905

LamarMoore, 2016-12-15:
Changes made to the code in #17905 have resulted in a performance regression. See stats here:

http://builds.mantidproject.org/view/Tests/job/master_performancetests/Master_branch_performance_tests/APITest.MatrixWorkspaceTestPerformance.test_move_bank_and_read_positions_SpectrumInfo_x10.htm

Anders-Markvardsen, 2017-01-09:
Will be targeted for 3.10

NickDraper, 2017-06-29:
Hard to see now how bad the loss of performance was as the graphs have moved on to the new baseline.

Haven't hear any specific comments on the log filtering performance from users of developers, so suggesting we accept the new baseline seeing as Tom left us.

Reopen if you disagree.

closed 2017-06-29 by NickDraper