mantidproject/mantid #18327: Masking by component in MaskDetectors

soininen, 2016-12-20:
Currently [`MaskDetectors`](http://docs.mantidproject.org/nightly/algorithms/MaskDetectors-v1.html) supports only index-based masking. There is, nowever, need, for component name based masking as well.

rosswhitfield, 2016-12-21:
Have a look at [MaskBTP](http://docs.mantidproject.org/nightly/algorithms/MaskBTP-v1.html). It doesn't work with component name by with bank/tube/pixel numbers, it might do what you want with a little work.

soininen, 2016-12-21:
@rosswhitfield Thanks for the tip! `MaskBTP` is too instrument-specific for our needs, however, although I might take another look at it with regards to the IN5 instrument at ILL. Implementing component masking in `MaskDetectors` didn't require that much effort, either, even though masking for example ranges of tubes like with `MaskBTP` in not possible.

PR #18343 (2017-01-11): Added masking-by-component to MaskDetectors
This PR adds a new *ComponentList* input property to [`MaskDetectors`](http://docs.mantidproject.org/nightly/algorithms/MaskDetectors-v1.html). The property takes a list of component names as strings and adds the detectors found recursively in the given components to the list of detector IDs to be masked.

Additionally, `ComponentCreationHelper::createTestInstrumentCylindrical()` was modified slightly to enable proper unit testing and the affected `InstrumentRayTracerTest` was updated.

**To test:**

Load your favorite dataset with attached instrumen [... cut]
closed 2017-01-11 by SimonHeybrock