mantidproject/mantid #18336: Provide 2nd version of LoadNXCanSAS and SaveNXcanSAS

AntonPiccardoSelg, 2016-12-21:
The NXcanSAS definition has been release and there are some changes to the attribute names. This means our loader and saver needs to be updated. In order to have backwards compatibility, we need to create slightly updated versions.

AntonPiccardoSelg, 2017-02-09:
NXcanSAS has been slightly modified by the CanSAS consortium. We need to create a second version of the loader.

We have to wait though until a schemer tool has been written by the consortium such that we can easily verify our work.

AntonPiccardoSelg, 2017-04-05:
Waiting for Steve to come back with news

AntonPiccardoSelg, 2017-10-05:
Ask Steve about the schemer

smk78, 2018-06-21:
Pete Jemian at APS has a validator on beta release: http://punx.readthedocs.io/en/latest/

smk78, 2018-06-22:
Follow up: punx may require

conda install lxml h5py numpy requests
pip install punx

prjemian, 2018-07-02:
The relevant documentation for NXcanSAS is here: http://download.nexusformat.org/doc/html/classes/applications/NXcanSAS.html

There are important differences from the draft version of the standard.

prjemian, 2018-07-02:
@smk78: punx requirements here: http://punx.readthedocs.io/en/latest/install.html#required-packages

gemmaguest, 2020-07-29:
Related to #28586

closed 2021-09-06 by stale[bot]