mantidproject/mantid #18358: Fedora 26 considerations

quantumsteve, 2017-01-04:
### Expected behavior

Continued limited support for the latest two Fedora releases. 

### Actual behavior

Two updates coming in Fedora 26 _may_ cause issues with Mantid. We should verify that Mantid supports the most recent version of Python and OpenSSL.

https://fedoraproject.org/wiki/Releases/26/ChangeSet#Python_3.6
https://fedoraproject.org/wiki/Releases/26/ChangeSet#OpenSSL_1.1.0

### Platforms affected

Fedora 26, Arch Linux, MacOS

quantumsteve, 2017-01-09:
I upgraded one of our OS X machines to Python 3.6 and now get the following error.

http://builds.mantidproject.org/job/master_clean-osx-10.11-python3

```
01:10:36 ======================================================================
01:10:36 ERROR: test_call_inside_function_uses_new_variable_name (__main__.SimpleAPITest)
01:10:36 ----------------------------------------------------------------------
01:10:36 Traceback (most recent call last):
01:10:36   File "/Users/builder/jenkins-linode/workspace/master_clean-osx-10.11-python3/Framework/PythonInterface/test/python/mantid/SimpleAPITest.py", line 228, in test_call_inside_function_uses_new_variable_name
01:10:36     raw = simpleapi.CreateWorkspace(DataX=dataX,DataY=dataY,NSpec=1)
01:10:36   File "/Users/builder/jenkins-linode/workspace/master_clean-osx-10.11-python3/build/bin/mantid/simpleapi.py", line 921, in algorithm_wrapper
01:10:36     lhs = _kernel.funcinspect.lhs_info(frame=frame)
01:10:36   File "/Users/builder/jenkins-linode/workspace/master_clean-osx-10.11-python3/build/bin/mantid/kernel/funcinspect.py", line 295, in lhs_info
01:10:36     ret_vals = process_frame(frame)
01:10:36   File "/Users/builder/jenkins-linode/workspace/master_clean-osx-10.11-python3/build/bin/mantid/kernel/funcinspect.py", line 183, in process_frame
01:10:36     ins_stack = decompile(frame.f_code)
01:10:36   File "/Users/builder/jenkins-linode/workspace/master_clean-osx-10.11-python3/build/bin/mantid/kernel/funcinspect.py", line 122, in decompile
01:10:36     i_arg_value = repr(code_object.co_consts[i_argument])
01:10:36 IndexError: tuple index out of range
01:10:36 
```

rosswhitfield, 2017-05-18:
I've fixed Arch Linux for python 3.6 (#18454) and OpenSSL 1.1.0 (#19432). I'm guessing this also fixed it for Fedora 26 and OSX? Can this be closed?

closed 2017-09-21 by quantumsteve