mantidproject/mantid #18376: Make WorkspaceGroup iterable

quantumsteve, 2017-01-08:
### Expected behavior

Mantid strives to have a clean, "pythonic" interface.

### Actual behavior

The solution to a [forum post](http://forum.mantidproject.org/t/sumspectra-of-grouped-workspace/200) required a ugly solution of iterating over a range starting at 1. 

### Potential solutions

The example above should look more like 

```python
#clone the first workspace as the result
wsResult = CloneWorkspace(wsGroup[0])
#loop over the rest if the members of the group
for ws in wsGroup:
    wsResult += ws
```

Modifying the `__getitem__` method to return an `IndexError` when the index is out of bounds,  would make the WorkspaceGroup iterable.

Alternatively, we could explicitly define an [`__iter__` method](http://www.boost.org/doc/libs/1_63_0/libs/python/doc/html/tutorial/tutorial/iterators.html),

AndreiSavici, 2017-01-10:
The above solution gives the wrong answer (it works). I think what @quantumsteve wants is:
#clone the first workspace as the result
```
wsResult = CloneWorkspace(wsGroup[0])
#loop over the rest if the members of the group
for ws in wsGroup[1:]:
    wsResult += ws
```
which yields:
```ArgumentError: Python argument types in
    WorkspaceGroup.__getitem__(WorkspaceGroup, slice)
did not match C++ signature:
    __getitem__(class Mantid::API::WorkspaceGroup {lvalue} self, unsigned __int64 index)
```

NickDraper, 2017-01-10:
Marked as low priority as this is an "it could be neater" issue from the dev team

closed 2020-07-29 by gemmaguest