mantidproject/mantid #18383: ApplyPaalmanPings unexpectedly modifies the can input workspace

soininen, 2017-01-09:
### Expected behavior
When running the script
```python
import numpy

xs = numpy.array([0.0, 1.0, 0.0, 1.1])
ys = numpy.array([2.2, 3.3])
sample_1 = CreateWorkspace(DataX=xs, DataY=ys, NSpec=2, UnitX='Wavelength')
ys = numpy.array([0.11, 0.22])
container_1 = CreateWorkspace(DataX=xs, DataY=ys, NSpec=2, UnitX='Wavelength')
corrected = ApplyPaalmanPingsCorrection(SampleWorkspace=sample_1, CanWorkspace=container_1)
```
one would expect the 'container_1' workspace to stay immutable over the call to `ApplyPaalmanPingsCorrection`.

### Actual behavior
`ApplyPaalmanPingsCorrection` will unconditionally run `RebinToWorkspace` on 'container_1' which, since the bins vary, will modify the binning of 'container_1'. This shouldn't happen as 'container_1' is an input property and as such should not be modified by the algorithm. Also, the modification will randomly delete the data behind the 'container_1' variable causing script failures.

SpencerHowells, 2017-01-10:
I started to write the PP routines about 10 yrs ago and they were then simple Py scripts. Since then we have been continually up-grading them to become proper algorithms and also adding extra functionality.
The original version was specifically for applying the corrections. The option to perform a simple subtraction was added against my wishes & advice. Subtraction only needs the Minus algorithm!
The corrections are in wavelength so the sample/container need converting from energy whereas a subtraction does not. So an error may have appeared during the modifications so that the incorrect workspace is used for subtraction. The input workspaces should not be modified.
Subtraction should only be performed on workspaces with identical energy/wavelength ranges for scientific reasons. In the early days (32bit?) I also had problems with rounding errors when performing Multiply/Divide. That is why there is a RebinToWorkspace. 
I would not be unhappy if the simple subtraction option is just removed. We have provided a user friendly interface to do this is Corrections>ContainerSubtraction.

Absorption corrections are still a work in progress. Martyn Giggs has re-written, with my advice, the Monte Carlo absorption algorithms. We have upgraded the algorithms used in Corrections>Absorption to use these & there is a Pull Request. The next stage is to write a PP version using these new algorithms. This will then replace the existing CalcPP which is my Fortran program from 40 yrs ago!

SpencerHowells, 2017-01-10:
In this section
def _subtract(self):
    """
    Do a simple container subtraction (when no corrections are given).
    """

    logger.information('Rebining container to ensure Minus')
    s_api.RebinToWorkspace(WorkspaceToRebin=self._can_ws_name,
                           WorkspaceToMatch=self._sample_ws_wavelength,
                           OutputWorkspace=self._can_ws_name)

    logger.information('Using simple container subtraction')

    # Rebin can
    s_api.RebinToWorkspace(WorkspaceToRebin=self._scaled_container_wavelength,
                           WorkspaceToMatch=self._sample_ws_wavelength,
                           OutputWorkspace=self._scaled_container_wavelength)

    s_api.Minus(LHSWorkspace=self._sample_ws_wavelength,
                RHSWorkspace=self._scaled_container_wavelength,
                OutputWorkspace=self._output_ws_name)

The first RebinToWorkspace is not needed.

soininen, 2017-01-10:
@SpencerHowells Thank a lot for the explanation! In my case converting the empty container to wavelengths just for the subtraction is also quite a lot of unnecessary work. So, I think in the end I'll actually do the subtraction manually. Thus eventually removing the 'empty container subtraction only' option from `ApplyPaalmanPingsCorrection` would not be a problem. However, it is out of the scope of this issue.

closed 2017-02-14 by martyngigg