mantidproject/mantid #18385: ApplyPaalmanPingCorrection fails if input workspace names do not contain underscores

soininen, 2017-01-09:
### Expected behavior
The algorithm should work with arbitrarily named workspaces.

### Actual behavior
If *SampleWorkspace* or *CanWorkspace* do not have and underscore in their names, the algorithm will raise `RuntimeError: substring not found`

### Steps to reproduce the behavior
```python
import numpy

xs = numpy.array([0.0, 1.0, 0.0, 1.1])
ys = numpy.array([2.2, 3.3])
sample = CreateWorkspace(DataX=xs, DataY=ys, NSpec=2, UnitX='Wavelength')
ys = numpy.array([0.11, 0.22])
container = CreateWorkspace(DataX=xs, DataY=ys, NSpec=2, UnitX='Wavelength')
corrected = ApplyPaalmanPingsCorrection(SampleWorkspace=sample, CanWorkspace=container)
```

SpencerHowells, 2017-01-10:
You are correct. These are changes that we have not done yet in our upgrading (see also my comment for 18383).

SpencerHowells, 2017-01-10:
I've found the errors:
1. 
at line 139
            can_cut = self._can_ws_name.index('_')    
            can_base = self._can_ws_name[:can_cut]
            prog_corr.report('Adding container filename')
            s_api.AddSampleLog(Workspace=self._output_ws_name,
                               LogName='container_filename',
                               LogType='String',
                               LogText=self._can_ws_name)
remove lines defining can_cut & can_base
for LogText replace can_base with self._can_ws_name
2.
at line 172
similar changes to
        sam_cut = self._sample_ws_name.index('_')
        sam_base = self._sample_ws_name[:sam_cut]
        prog_wrkflow.report('Adding sample filename')
        s_api.AddSampleLog(Workspace=self._output_ws_name,
                           LogName='sample_filename',
                           LogType='String',
                           LogText=sam_base)

closed 2021-03-02 by stale[bot]