mantidproject/mantid #18399: Reflectometry interface (Polref): make processing instructions work properly

raquelalvarezbanos, 2017-01-10:
This issue is similar to https://github.com/mantidproject/mantid/issues/18208.

### Expected behavior

Users should be able to enter processing instructions like this: `3, 4` in the `Settings` tab.

### Actual behavior

An error message is shown (`Invalid key value pair, '4'`) which is not very useful.

### Steps to reproduce the behavior

Open Reflectometry (Polref) interface, set instrument to `INTER` and enter the following data in the table:
- Run(s): 13460
- Angle: 0.7
Then go to `Settings` and enter `3, 4` in `ProcessingInstructions`.

### Platforms affected

All.

PranavBahuguna, 2017-01-11:
Should I apply this to all other quoted setting values (such as direct beam)?

raquelalvarezbanos, 2017-01-11:
Yes, similar properties will have the same problem.

PR #18413 (2017-01-16): Reflectometry Interface (Polref): Allow setting values to be passed without enclosing quotes
This is relatively similar to PR #18253. This allows the 'Direct Beam' and 'Processing Instructions' settings values in the interface to be passed without enclosing them in quote marks.

**To test:**

- Open the ISIS Reflectometry (Polref) interface.
- Import INTER_NR_test2.tbl (can be obtain from sample datasets).
- Go to the 'Settings' tab and enter unquoted values into 'Processing Instructions' (e.g. `3, 4`).
- Select some runs in the 'Runs' tab and process - it should not throw an error regarding t [... cut]
closed 2017-01-16 by SimonHeybrock