mantidproject/mantid #18438: SplineInterpolation not copying Instrument correctly

samueljackson92, 2017-01-16:
### Expected behavior
Output workspace should match the input workspace instrument. In particular the detector IDs for each spectrum should match.

### Actual behavior
@VickieLynch noticed that the SplineInterpolation algorithm gives her an output workspace where the detector IDs for each spectra don't match the input.

### Steps to reproduce the behavior
This should reproduce the issue:
```python
Vickie Lynch [1:36 PM] 
spec=LoadNexus(Filename='spec.nxs')
denom=LoadNexus(Filename='denom.nxs')
SplineInterpolation(WorkspaceToMatch='denom', WorkspaceToInterpolate='spec', OutputWorkspace='spec_interp')
interp=Divide(LHSWorkspace='spec_interp', RHSWorkspace='denom', OutputWorkspace='newSpec')
#for i in range(spec.getNumberHistograms()):
#   sp=spec.getSpectrum(i)
#   sp2=interp.getSpectrum(i)
#   ids = sp.getDetectorIDs()
#   sp2.setDetectorID(ids[32832])
ConvertUnits(InputWorkspace='newSpec', OutputWorkspace='newSpec', Target='TOF')
```

### Platforms affected
All

reimundILL, 2017-01-16:
Hi Samuel, this may be unrelated? I found a bug concerning the OutputWorkspace not being correctly set and I fixed it in PR
https://github.com/mantidproject/mantid/pull/18138/
Maybe this helps, commit 2f59d70 runs correctly (I fixed a wrong logging message later). I fixed the `setupOutputWorkspace` function.

samueljackson92, 2017-01-16:
@reimundILL Possibly. I'll run @VickieLynch's script with your PR and see what happens.

samueljackson92, 2017-01-16:
This was fixed in this [commit](https://github.com/mantidproject/mantid/pull/18138/commits/d84c2e2ff73dcea3b286bf6c72dc8e7dac13dd4d) on in #18138. As such this should be closed.

commit d84c2e2f (2017-01-16): Refs #18097 Create output workspace by cloning.

This will ensure that the vertical axis is copied as before, but will
also ensure the correct detector mapping is also retained.

As this is fixed here this closes #18438.
closed 2017-01-16 by samueljackson92