mantidproject/mantid #18442: Is `CalculateResolution` calculating theta properly?

raquelalvarezbanos, 2017-01-16:
If I run this script:

```
ws = Load('INTER00013460')
res, two_theta = CalculateResolution(Workspace = ws)
print res, two_theta
```

the output is:

```
0.0340291462898 0.700020432472
```

which according to the documentation means that `TwoTheta = 0.7`. However I know that for this run `Theta=0.7`, which can be seen in the sample logs.

This issue is to clarify what is happening, but having a quick look it seems that the algorithm is getting theta from the logs and using it as `2 * theta`.

raquelalvarezbanos, 2017-01-30:
In addition to the above, scientists have an excel spreadsheet they use to calculate the resolution: 
[FullResolution_Calculator_BW_CJK.xlsx](https://github.com/mantidproject/mantid/files/738613/FullResolution_Calculator_BW_CJK.xlsx). I checked with Jos the equation the are using for the resolution and the equation used in `CalculateResolution` and they seem to be different. We also need to clarify which one is the correct one, or if they are equivalent ways of calculating the same thing.

raquelalvarezbanos, 2017-02-13:
In addition, scientists want to rename this algorithm to `NRSlitResolution`. We also should remove output property `ThetaOut` which is not needed/used.

raquelalvarezbanos, 2017-02-14:
Latest excel spreadsheet sent by Christy: 
[Copy of FullResolution Calculator_BW_CJK (2).xlsx](https://github.com/mantidproject/mantid/files/774528/Copy.of.FullResolution.Calculator_BW_CJK.2.xlsx)

gemmaguest, 2017-09-11:
To clarify what is happening with the original question about the algorithm returning `theta` instead of `twoTheta`, this is because the default log name for the `TwoTheta` property is `Theta`. If you were to override this to provide a more sensible log name for `TwoTheta` then it would work as described. This default should be changed/removed because it is confusing. We should perhaps have an option to specify the log name for `Theta` instead/as well - I'll ask the scientists what they would prefer.

PR #20450 (2017-09-19): Rename and add fixes to CalculateResolution
***Note*** *The branch name and commits reference the wrong issue (#18842 instead of #18442)*

The following changes have been made to `CalculateResolution`:

- The algorithm has been renamed to `NRCalculateSlitResolution` as this algorithm is specific to neutron reflectometry, and the resolution it calculates is the slit resolution.
- The `TwoThetaLogName` property has been replaced by `ThetaLogName`. This still takes `Theta` as the default value, which was causing confusion before because it was being used as two theta. Is is now being used as thet [... cut]
closed 2017-09-19 by martyngigg