mantidproject/mantid #18457: ComputeCalibrationCoefVan uses hard-coded sample log entry for temperature

soininen, 2017-01-17:
The [`ComputeCalibrationCoefVan`](http://docs.mantidproject.org/nightly/algorithms/ComputeCalibrationCoefVan-v1.html) algorithm reads the temperature information from 'temperature' sample log entry which is specific to MLZ. The correct temperature log name could rather be specified in the instrument parameters file to make the algorithm more general.

AndreiSavici, 2017-01-17:
I suggest allowing users to input this either as a number (in case log is missing or bad), or as a log name. See for example http://docs.mantidproject.org/nightly/algorithms/ApplyDetailedBalance-v1.html

soininen, 2017-01-18:
@AndreiSavici Thanks for the feedback! There is a pull request #18433  for user-inputable temperature already, although it only takes a number as an argument. We were thinking about using the instrument parameters file to name the corresponding sample log entry as its name is (usually) instrument specific and shouldn't change too often.