mantidproject/mantid #18461: Scope out what needs to be done to remove vms_compat

AntonPiccardoSelg, 2017-01-18:
`vms_compat` seems to be a legacy system. We need to ask Freddie how to get rid of it for more modern loaders.

Files which use vms_compat:

* LoadEventNexus.cpp
* SaveISISNexus.cpp
* UpdateInstrumentFromFile.cpp
* LoadIISISNexus2.cpp
* CreateSimulationWorkspace.cpp
* LoadNexusMonitors2.cpp
* LoadISISNexus2.h
* SaveISISNexus.h

| ----

FreddieAkeroyd, 2017-01-25:
I will write you a table of the equivalents from elsewhere in the file

AntonPiccardoSelg, 2017-01-26:
@FreddieAkeroyd  Thank you!

AntonPiccardoSelg, 2017-10-05:
@NickDraper I assume this might be become relevant soon. Therefore I have assigned this to you instead of a simple unassign

NickDraper, 2020-01-10:
@StephenSmith25 This might be helpful to see the bigger picture

closed 2020-07-24 by NickDraper