mantidproject/mantid #18470: New EFixed mode needed for MonteCarloAbsorption

gvardany, 2017-01-19:


martyngigg, 2017-01-19:
@gvardany This is good. Spencer was talking about this and it will help avoid the hack we would currently have to do to get this working.

I'd be happy to test this once it's implemented.

SpencerHowells, 2017-01-19:
This is a reasonable approximation for QENS mode and the existing version works.
The problem is in FixedWindow mode where the data has x-Unit of Q. The MC absorption algo requires wavelength and detector angle. So the workspace which is a single hist with x in Q with nbins values has to transformed to  ws with nbins hists with a single x value of wavelength. The hists should also have the detector angle. The algo should then work.

gvardany, 2017-01-20:
@martyngigg I was wondering what will happen if one gives `NumberOfWavelengthPoints=1`.
Looking from the code, it seems it will take the lower edge of the spectrums x-axis.
If it took the middle point, normally corresponding to energy transfer = 0 (if the spectrum is symmetric of course), then, in principle, one could achieve the desired behaviour just by number of wavelength points of 1, instead of the need of new option. Even if it is not symmetric around E=0, it is more correct to approximate to the lambda corresponding to the middle point rather than lambda_fixed. 
Regarding the wavelength unit requirement, one possibility is:
if it is not in wavelength, neither anything convertible to it, attempt to take the lambda_fixed from IDF, and  set `NumberOfWavelengthPoints=1`.
Like this it seems we can cover all the use cases, without really changing the signature of the algorithm.

gvardany, 2017-01-20:
@SpencerHowells Indeed this is for fixed window scans. For us currently, the x-axis is the temperature (or any other log), while spectrum axis is linked to detectors. For your case transposing the workspace seems necessary anyway, but the creation of a workspace with single x-value of wavelength is really a workaround, that will be solved by this.

martyngigg, 2017-01-20:
@gvardany It should take the mid-point of any bin. It calls `outputWS->points(i)` which will give you back the mid point of the bins or the point itself depending on whether it is a histogram or point data.

It would certainly be good to relax the wavelength restriction and at least allow it to internally convert to wavelength if it can. In some ways I'd be hesitant though to make it too silent that behaviour may have changed if you select `EMode`=`Indirect` and then it ends up going to do something else, I'd think I'd actually prefer if to be explicit that you've requested that.

SpencerHowells, 2017-01-20:
Fixed window is Elastic & use the single wavelength input.

gvardany, 2017-01-20:
Ok, it's good that it takes the mid points. 

My point was that, unlike PaalmanPings, where we have the option Elastic/Inelastic/EFixed as input property, here it is fully automated. So maybe it is not nice to add a property just for one option. From  the other hand the mode is deduced from the instrument, but we need to be able to use different modes for the same instrument. So that's why I thought it could be nice to automatise also the EFixed behaviour somehow. Whereas in PaalmanPings, of course we can add the option explicitly.

@SpencerHowells Exactly, and this can be computed from EFixed. Actually our instrument can do also inelastic fixed window scan, however, again the energy kick is negligible compared to Efixed. For QENS, it is normally negligible, but can in principle be non-negligible, so we need to have an option, but it can be automatised as I mentioned above.

gvardany, 2017-01-23:
@martyngigg So should we add a user property with 4 options, similar to PaalmanPings algorithms, or make it automatic also for EFixed?

martyngigg, 2017-01-23:
@gvardany I think it's probably best to be consistent with what is already there and keep it automatic then. For some reason I thought we already had the `EMode` property. Those can always be added in the future should we feel the need to to explicitly override things at the algorithm input level.

SpencerHowells, 2017-01-23:
I need to look at both my codes - both Py alg & Fortran routine - to remind myself why Efixed is needed as input. 2 things spring to mind: the routines work for diffraction as well as inelastic; flat plate does not do a proper inelastic correction.

SpencerHowells, 2017-01-24:
Having emode/efixed as input in PaalmanPings is left over from a time long ago when I did not know how to read it from the ws. 
Emode is needed so that it can use Elastic for the Diffraction input.
FlatPlatePP does not need efixed.

gvardany, 2017-02-01:
I have opened/implemented #18652 for flat plate and cylinder paalman pings corrections. 
I am still thinking to add the options explicitly to MonteCarloAbsorption. It seems not right to decide on the workspace unit, since it can be not in wavelength, but e.g. in energy transfer, and the user might expect it to behave differently. It also seems nice to have consistent interfaces between monte carlo and paalman pings. I will open a separate ticket for MonteCarlo one.

SpencerHowells, 2017-02-01:
Absorption corrections are a function of wavelength so it is logical that the input to the Absorption algo is in wavelength. It can then be used by anyone. It is then up to the instrument algos to ensure that the input is correct.
I have not updated the PP interfaces - eventually it will use the MonteCarlo method.
I have also started rewriting the code to produce individual MC algos for FlatPlate, Cylinder & Annulus which are then called from the Indirect versions & the interface.

gvardany, 2017-02-01:
Yes the correction is a function of wavelength, these extensions will just make them easier to use if one needs the correction factors only for a single wavelength read from analyser. 
Do you mean that MonteCarlo algorithm will also compute As,sc, and Ac,sc factors in future?

martyngigg, 2017-02-01:
@gvardany I think I'm inclined to agree with Spencer here in that the algorithm should just expect wavelength input and client's responsibility to ensure the input is in the correct form. If we allow it for this one case to be different then I'm not sure where we end up stopping if others come along in the future. As it stands with wavelength as input it's consistent with all other absorption correction algorithms so they are all easy to compare. In this case I expect this will all be wrapped up in code that most users won't have to deal with? 

For PP there will likely be a new algorithm to calculate the factors using `MonteCarloAbsorption` to do the heavy lifting.

gvardany, 2017-02-01:
I see your point. We have re-discussed this again here. Still the alternative options seems less elegant. The problem again, is that for FWS data, we have a workspace with 2theta as y-axis and temperature as x-axis, and they correspond to lambda_in = lambda_out = lambda_fixed derived from the analyser energy. The alternative solution is:
- To create an algorithm that creates a fake workspace with the same 2theta y-axis and single wavelength. 
- Invoke this algorithm in Indirect->Corrections, if the instrument is ILL, to be able to compute the corrections, and then apply the corrections to the original workspace.
The PR #18653 still seems to us less intrusive. It is absolutely transparent for the existing use, also in terms of the unit requirement, just has one more option that behaves differently.

So how should we proceed? Any other suggestion?

martyngigg, 2017-02-01:
I don't quite see how #18563 relates to `MontCarloAbsortion`?

gvardany, 2017-02-01:
PR #18653 is for non-MC paalman pings corrections. Btw are they going to be deprecated when the Monte Carlo ones are created? But eventually we would need similar behaviour in MonteCarloAbsorption, and all the `Indirect` wrappers around it. However, MonteCarloAbsorption is different in a sense, that Elastic/Direct/Indirect it decides automatically, so does not have any option. Will the future paalman pings be also like that?

SpencerHowells, 2017-02-01:
Paalman Pings will be deprecated. It suffers from many faults. FlatPlate does not do Inelastic & treats it as Elastic. Annular sample is not done properly, it only does the outer wall of the can.

martyngigg, 2017-02-01:
For `MonteCarloAbsorption` I don't think we want to do the conversion internally because if I need to run this multiple times, in the case of Paalman & Pings then I don't want to keep doing the conversion each time. I'd rather know that my input is in wavelength and I can just carry on.

gvardany, 2017-02-02:
OK, I agree that it should not be in MonteCarloAbsorption itself. Then we could do the placeholder workspace  creation either in the GUI or in the indirect wrappers around MC absorption, or have ill-specific wrappers around them. Where it would be the best? It would be nicer though to have in the algorithms, so that one can use them also from python directly.

SpencerHowells, 2017-02-02:
(OS)IRIS also do pseudo fixed window scans and produce workspaces which are Q v temperature rather than angle v temperature. So it would be nice to have a wrapper algo which we could both use.

SpencerHowells, 2017-02-02:
#18662 creates MC equivalents of the existing absorption algos for the 3 standard geometries. They set the geom, material & beam & run MCabsorption. They would remove repetitive code in other algos and be easier for users.

gvardany, 2017-02-02:
I think MC equivalents of the absorption algos have already been merged in PR #18282 .
The new issue, as far as I understand, is to separate the computation and applying of the corrections, which is good.
But in the algorithms that compute the corrections, one new option can be added for FWS.
E.g. for flat plate absorption, I see that only lines https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/IndirectFlatPlateAbsorption2.py#L131-L137 
that should be different for FWS.
When applying the correction there should be no difference in principle. Even the FWS corrections will have one bin, both the Minus and Divide algorithms will run just fine.

SpencerHowells, 2017-02-02:
The algos in #18282 will use the 3 new algos cutting down on repetitive coding so will also need modifying. Louise already has the new versions.
For the lines you mention l'm already working ( slowly) on a version with options - so far for Inelastic & Diffraction. We could add another for FWS.

gvardany, 2017-10-03:
Thanks to PR [#20373](https://github.com/mantidproject/mantid/pull/20373), this issue is no longer relevant.

closed 2017-10-03 by gvardany