mantidproject/mantid #18477: Script giving different results in Mantid v3.8 and a recent branch

raquelalvarezbanos, 2017-01-19:
The following script:

```
Load(Filename=r'\\isis\inst$\ndxinter\instrument\data\cycle_16_2\INTER00038415.nxs', OutputWorkspace='38415', LoadMonitors=True)
FilterByTime(InputWorkspace='38415', OutputWorkspace='38415_filter', StopTime=200)
Rebin(InputWorkspace='38415_filter', OutputWorkspace='38415_0_200', Params='0,100,100000', PreserveEvents=False)
```

is giving me different results on Mantid 3.8 and a recent branch.

raquelalvarezbanos, 2017-01-19:
Mantid 3.8:

[image]

This branch (debug mode, after fix for `LoadEventNexus`, see commit above):

[image]

raquelalvarezbanos, 2017-01-19:
@NickDraper @martyngigg

raquelalvarezbanos, 2017-01-19:
`Rebin` doesn't seem to be the problem, the following script:

```
Load(Filename=r'\\isis\inst$\ndxinter\instrument\data\cycle_16_2\INTER00038415.nxs', OutputWorkspace='38415', LoadMonitors=True)
Rebin(InputWorkspace='38415', OutputWorkspace='38415_0_200', Params='0,100,100000', PreserveEvents=False)
```

gives the same results on 3.8 and this branch:

[image]

raquelalvarezbanos, 2017-01-20:
The script below (where I've replaced `PreserveEvents=False` -> `PreserveEvents=True` in `Rebin`):

```
Load(Filename=r'\\isis\inst$\ndxinter\instrument\data\cycle_16_2\INTER00038415.nxs', OutputWorkspace='38415', LoadMonitors=True)
FilterByTime(InputWorkspace='38415', OutputWorkspace='38415_filter', StopTime=200)
Rebin(InputWorkspace='38415_filter', OutputWorkspace='38415_0_200', Params='0,100,100000', PreserveEvents=True)

ws = mtd['38415_0_200']

import numpy as np

event_list = ws.getEventList(0)
data = event_list.getTofs()
counts, bins = np.histogram(data, bins=range(0, 100000, 100))
ws2 = CreateWorkspace(bins, counts)
```

gives this graph:

[image]

which is identical to what I see on release 3.8, so after rebinning the events seem to be there in the workspace, but I can't see them when I plot workspace `38415_0_200` or when I open the matrix view.

martyngigg, 2017-01-20:
Thanks, @raquelalvarezbanos.

If the events are there then my assumption would be that there is something wrong with the rebinning in a very subtle manner. Are you still looking into it or would you like me to take a look?

raquelalvarezbanos, 2017-01-20:
Thanks @martyngigg I am looking at it, I may ask you for help at some point if I get stuck. I'll keep adding comments as I discover new things.

raquelalvarezbanos, 2017-01-20:
Added a new line to the last script:

```
Load(Filename=r'\\isis\inst$\ndxinter\instrument\data\cycle_16_2\INTER00038415.nxs', OutputWorkspace='38415', LoadMonitors=True)
FilterByTime(InputWorkspace='38415', OutputWorkspace='38415_filter', StopTime=200)
Rebin(InputWorkspace='38415_filter', OutputWorkspace='38415_0_200', Params='0,100,100000', PreserveEvents=True)

ws = mtd['38415_0_200']

import numpy as np

event_list = ws.getEventList(0)
data = event_list.getTofs()
counts, bins = np.histogram(data, bins=range(0, 100000, 100))
ws2 = CreateWorkspace(bins, counts)

ws3 = CreateWorkspace(ws.readX(0), ws.readY(0))
```

Below is a comparison of workspaces `ws2` and `ws3`:

[image]

[image]

Both of them were created from `ws` (the workspace resulting from `Rebin`), but `ws2` was created using the event list while `ws3` was created using the method `readY()`.

peterfpeterson, 2017-01-20:
I think that [`EventList::sortTof()`](https://github.com/mantidproject/mantid/blob/c4beafbeb9146de435cc6f78ec05249976ddb60b/Framework/DataObjects/src/EventList.cpp#L1102) could have a check to see if the `EventList` is empty and just change the sorting and skipping the work. An empty list is always sorted the way you want it.

Since `FilterByTime` configures a call `FilterEvents`, the error with the sorting is hiding in there. It passes the responsibility down the chain to `EventList` which must have an issue with keeping the events in the correct order when filtering. I didn't see the bug in `splitByTime*Helper` (multiple methods with similar names), but it must be in there since it is the only place where events are copied from the original list to the new one.

wdzhou, 2017-01-20:
I agree with Pete.  I traced it back to method void EventList::splitByFullTimeHelper().  It works based on the assumption that all the events are ordered by absolute time.

PR #18504 (2017-01-24): Ensure output EventLists have correct sort order after filtering
Description of work.

Fixes an issue with `EventList` where the sort order was inconsistently defined after `FilterByTime`.

**To test:**

The following scripts creates a filtered event workspace:
```
ws = CreateSampleWorkspace("Event",BankPixelWidth=1)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:00:00", Value=100) 
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:10:00", Value=100)
AddTimeSeriesLog(ws, Name="proton_charge", Time="2010-01-01T00:20:00", Value=100)
AddTimeSeriesLog(ws, Name="proton_ [... cut]
closed 2017-01-24 by AndreiSavici