mantidproject/mantid #18491: Mantid won't build with SIP 4.19

quantumsteve, 2017-01-20:
### Expected behavior

### Actual behavior

```
[3/351] Building CXX object MantidQt/Python/CMakeFiles/mantidqtpython.dir/sip_mantidqt.cpp.o
FAILED: MantidQt/Python/CMakeFiles/mantidqtpython.dir/sip_mantidqt.cpp.o 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DBOOST_ALL_DYN_LINK -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_POSIX_TIME_STD_CONFIG -DENABLE_OPENCASCADE -DHAVE_STDINT_H -DMAKE_VATES -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_XML_LIB -Dmantidqtpython_EXPORTS -isystem /usr/local/include -isystem /usr/local/include/nexus -I/Users/svh/Documents/MantidProject/mantid/Framework/Kernel/inc -I/Users/svh/Documents/MantidProject/mantid/Framework/HistogramData/inc -I/Users/svh/Documents/MantidProject/mantid/Framework/Indexing/inc -I/Users/svh/Documents/MantidProject/mantid/Framework/Geometry/inc -I/Users/svh/Documents/MantidProject/mantid/Framework/API/inc -isystem /usr/local/Cellar/qt/4.8.7_2/include -iframework /usr/local/Cellar/qt/4.8.7_2/lib -isystem /usr/local/Cellar/qt/4.8.7_2/include/QtOpenGL -isystem /usr/local/Cellar/qt/4.8.7_2/include/QtSvg -isystem /usr/local/Cellar/qt/4.8.7_2/include/QtGui -isystem /usr/local/Cellar/qt/4.8.7_2/include/QtXml -isystem /usr/local/Cellar/qt/4.8.7_2/lib/QtCore.framework/Headers -I/Users/svh/Documents/MantidProject/mantid/MantidQt/API/inc -IMantidQt/API -I/Users/svh/Documents/MantidProject/mantid/MantidQt/MantidWidgets/inc -IMantidQt/MantidWidgets -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../SliceViewer/inc -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../MantidWidgets/inc -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../MantidWidgets/inc/MantidQtMantidWidgets -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../MantidWidgets/inc/MantidQtMantidWidgets/InstrumentView -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../Factory/inc -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../SpectrumViewer/inc -I/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../RefDetectorViewer/inc -IMantidQt/Python/../SliceViewer -IMantidQt/Python/../RefDetectorViewer -Wall -Wextra -Wconversion -Winit-self -Wpointer-arith -Wcast-qual -Wcast-align -fno-common -Wno-deprecated -Wno-write-strings -Wno-sign-conversion -Woverloaded-virtual -fno-operator-names -O2 -g -DNDEBUG -fPIC   -std=gnu++14 -MD -MT MantidQt/Python/CMakeFiles/mantidqtpython.dir/sip_mantidqt.cpp.o -MF MantidQt/Python/CMakeFiles/mantidqtpython.dir/sip_mantidqt.cpp.o.d -o MantidQt/Python/CMakeFiles/mantidqtpython.dir/sip_mantidqt.cpp.o -c /Users/svh/Documents/MantidProject/build/MantidQt/Python/sip_mantidqt.cpp
In file included from /Users/svh/Documents/MantidProject/build/MantidQt/Python/sip_mantidqt.cpp:18:
/Users/svh/Documents/MantidProject/build/MantidQt/Python/sipmantidqtpythonpart0.cpp:16365:27: error: call to implicitly-deleted copy constructor of '::MantidQt::Factory::WidgetFactory'
            sipCpp = new  ::MantidQt::Factory::WidgetFactory(*a0);
                          ^                                  ~~~
/Users/svh/Documents/MantidProject/mantid/MantidQt/Python/../Factory/inc/MantidQtFactory/WidgetFactory.h:48:51: note: copy constructor of 'WidgetFactory' is implicitly deleted because base class 'QObject' has an inaccessible copy constructor
class EXPORT_OPT_MANTIDQT_FACTORY WidgetFactory : public QObject {
                                                  ^
1 error generated.
ninja: build stopped: subcommand failed.
```

### Steps to reproduce the behavior

### Platforms affected

Homebrew already updated to SIP 4.19. Other distributions will likely follow soon.

AndreiSavici, 2017-01-20:
Ubuntu 16.10, and for now 17.04, are still on SIP 4.18

rosswhitfield, 2017-01-20:
Fedora Rawhide is 4.19

PR #18492 (2017-01-23): Fix building with sip 4.19
Description of work.

This updates deprecated sip functions that were removed in sip 4.19. 

After installing SIP 4.19, I also had to upgrade to PyQt 4.12. Previous version fail to build.

**To test:**

Fixes #18491.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is  [... cut]
closed 2017-01-23 by SimonHeybrock