mantidproject/mantid #18548: Logging locks up Algorithms

OwenArnold, 2017-01-25:
### Steps to reproduce the behaviour

Run  this with Logging set to `Error` in Mantid
``` python
import numpy

yvalues = np.zeros(5*512*512)
evalues = np.zeros(5*512*512)
xvalues = np.zeros(5)

ws = CreateWorkspace(DataX=xvalues, DataY=yvalues, DataE=evalues, NSpec=512*512)
```
Now re-run that script with logging set to `Debug`. This runs significantly slower owing to the logging of the input properties.

I think this could really trip people up. Especially for those importing data from other bits of software. Figuring out what's going wrong will not be easy as currently stands for our users.

The suggested fix for this is to abbreviate the logged arrays for `FloatArrayProperty`, `IntArrayProperty` (possibly any array input type) when the array exceeds a maximum size to be something like `[0,1,2,...97,98,99]`. The History must record the actual values. Making the size configurable would be a good idea. A default of 100 items before abbreviation, is probably sensible.

@martyngigg Has anything like this been considered already.

martyngigg, 2017-01-25:
@OwenArnold This is a known issue and has been for some time but no users have every really complained so it has not been pushed up the priority list: #11317

martyngigg, 2017-01-25:
I actually didn't realise that issue was mine :man_facepalming:

OwenArnold, 2017-01-25:
As this affects `CreateWorkspace` I think it is potentially a little more serious that #11317 highlights. This algorithm is a major part of our python compatibility layer. I'm also a little concerned that the users would not easily identify the issue should it occur. Having said that, it's been around for a while and does not need to be part of release 3.9 in my opinion. @martyngigg What do you think about the suggested fix?

martyngigg, 2017-01-25:
I think part of the reason I didn't address it at the time was that it wasn't obvious what the fix is. To some extent this will happen at some point when you're just trying to dump a lot of information to the log window in the main thread. I had though that some type of buffering might be the way to go but it's not clear.

OwenArnold, 2017-01-25:
Given that workspace have history. I do not think that it's necessary for the Logs to provide a way to be runnable. I think the solution is to abbreviate what gets written to the logs to reduce the volume.

martyngigg, 2017-01-25:
True. A fix for this case would be to abbreviate log printing like `numpy` does. This probably reduces the need to address the more general problem of fast, high-volume logging.

samueljackson92, 2019-01-29:
Transferring to @OwenArnold who reported this. But it might make a good student/new starter issue once they get going?

closed 2021-03-02 by stale[bot]