mantidproject/mantid #18550: Update scripts for compatibility with both Python 2 & 3

rosswhitfield, 2017-01-25:
A cheat sheet is available at http://python-future.org/compatible_idioms.html that details the differences between Python 2 & 3 code. Most of our incompatibilities fall in to one of three categories:
- `print` is now a function and requires parentheses in Python 3
- commas not allowed in `except` clauses.
- `dict.iteritems()` no longer exists 

We need to maintain compatibility with Python 2. See the [Python 3](http://www.mantidproject.org/Python_3) page on the wiki for more detail. You should build mantid with python3 in order to test your changes.

I've tried to grouped the files where it makes sense but feel free to move them around if required. You should also look at and fix any systemtests that relate to the files you are changing.

This is an umbrella issue. Please add your username to the list below and include this ticket number in your PR.
1.
2.
3.
4.
5.
6.
7.
8.
9.
10.
11.
12.
13.
14.
15.
16.
17.
18.
19.
20.
21.
22.

**Group 1**
```
scripts/TofConverter/convertUnits.py
scripts/TofConverter/converterGUI.py
scripts/TofConverter/__init__.py
scripts/QECoverage.py
scripts/TofConverter.py
scripts/FilterEvents/eventFilterGUI.py
scripts/FilterEvents/MplFigureCanvas.py
scripts/FilterEvents/__init__.py
scripts/FilterEvents.py
```

**Group 2**
```
scripts/DGSPlanner.py
scripts/DGSPlanner/DimensionSelectorWidget.py
scripts/DGSPlanner/MatrixUBInputWidget.py
scripts/DGSPlanner/ClassicUBInputWidget.py
scripts/DGSPlanner/InstrumentSetupWidget.py
scripts/DGSPlanner/ValidateOL.py
scripts/DGSPlanner/__init__.py
scripts/DGSPlanner/DGSPlannerGUI.py
```

**Group 3**
```
scripts/SANS/reduction_settings.py
scripts/SANS/sans/state/state_functions.py
scripts/SANS/sans/state/slice_event.py
scripts/SANS/sans/state/data.py
scripts/SANS/sans/state/state.py
scripts/SANS/sans/state/state_base.py
scripts/SANS/sans/state/reduction_mode.py
scripts/SANS/sans/state/wavelength.py
scripts/SANS/sans/state/normalize_to_monitor.py
scripts/SANS/sans/state/automatic_setters.py
[... 426 lines cut ...]
scripts/Examples/SampleLogs_Demo.py
scripts/Examples/HRPDDialog_Example.py
scripts/Examples/TableWorkspace_Example.py
scripts/Examples/StripPeaks.py
```

**Misc, to be grouped (move them to your group if needed)**
```
scripts/ORNL_SANS.py
scripts/REFL_SF_Calculator.py
scripts/Engineering/EnggUtils.py
scripts/reducer_singleton.py
scripts/TemporaryREF_MScripts/testCenterREF_M.py
scripts/MantidIPython/plot_functions.py
scripts/MantidIPython/__init__.py
scripts/DGS_Reduction.py
scripts/REFM_Reduction.py
scripts/REFL_Reduction.py
scripts/HFIR_4Circle_Reduction.py
scripts/Powder_Diffraction_Reduction.py
scripts/ISIS_Reflectometry.py
scripts/PearlPowderISIS/pearl_routines.py
scripts/DiamondAttenuationCorrection/UBMatrixGenerator.py
scripts/DiamondAttenuationCorrection/FitTrans.py
```

DimitarTasev, 2017-02-17:
Group 7 will be completely removed and restructured by #18379

peterfpeterson, 2017-06-22:
Group 7 doesn't appear to exist any more. Can it be marked as "done"?

rosswhitfield, 2017-06-22:
Sure

commit b9555ada (2017-08-30): Fix misc python 3 script stuff

Closes #18550
closed 2017-08-30 by AndreiSavici