mantidproject/mantid #18551: FilterEvents gui can have backwards time

AnthonyLim23, 2017-01-25:
### Expected behavior
All of the sample logs to produce plots where time always progresses forwards.  
### Actual behavior
For some cases the line drawn in the plot window is go backwards in time (see image). 
[image]

### Steps to reproduce the behavior
I have used 
[CNCS_7860_event.zip](https://github.com/mantidproject/mantid/files/730091/CNCS_7860_event.zip) and the sample log phase1. I have also extracted the raw data from the file and got the blue line in the figure. Clearly the time always increases and the times are smaller in the raw data. 
### Platforms affected
Windows 7.

NickDraper, 2017-02-14:
I can't reproduce this with the same data and log file with 3.9.0
[image]

even when changing the line type to line rather than horizontal steps
[image]

Is there anything else that needs to be done to reproduce this?

AnthonyLim23, 2017-02-14:
I cant reproduce it either with V3.8

closed 2017-02-16 by NickDraper