mantidproject/mantid #18553: TofConverter cannot convert to Nu(Thz)

AnthonyLim23, 2017-01-25:
### Expected behavior
If the 'Convert To:' has 'Nu(Thz)' selected, then when convert is pressed a number should appear in the converted value box. 
### Actual behavior
For all input units (including Nu(Thz)) the error message
 'local variable OutputVal referenced before assignment' 
appears in a pop-up box, if  'Convert To:' is set to Nu(Thz). 

Input units of Nu(Thz) and any other option for 'Convert To:' produce sensible results. 
### Steps to reproduce the behavior
Enter an input value and select the units (any will do). In the 'Convert To:' box select Nu(Thz) and fill in any other relevant boxes. Then press convert and the pop-up box will appear with the error message. 
### Platforms affected
Windows 7.

NickDraper, 2017-06-29:
Add error message for unsupported units, or event better remove them from the list

KarlPalmen, 2017-07-20:
I found out that TofConverter is a utility interface.

PR #20012 (2017-07-21): TofConverter fixed to convert to NuThz
Changed case of character to fix TofConverter bug

**To test:**
See the issue

Fixes #18553 

**Release Notes** 
*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [ ] If there are changes in th [... cut]
closed 2017-07-21 by ianbush