mantidproject/mantid #18566: Muon Analysis fails after change of instrument and fitting function

KarlPalmen, 2017-01-26:
### Expected behaviour
In the muon analysis 'unscripted' tests group 2, in test 7, one should be able to fit `LinearBackground`. 

### Actual behavior
Mantid throws an unexpected exception complaining about a non-existent parameter.

### Steps to reproduce the behaviour
Run test 1 then test 7 in 
http://docs.mantidproject.org/nightly/interfaces/MuonAnalysis_test_guides/Muon_Analysis_test_group_2_fitting.html#muon-analysis-testguide-2-fitting-ref

### Platforms affected
Window 7 at least

raquelalvarezbanos, 2017-01-26:
@KarlPalmen Could you double check this? I am not able to reproduce this bug.

KarlPalmen, 2017-01-26:
It works OK on a recent build I used for test pull requests. I think you need to download the nightly build required for the unscripted testing. I'll try it again with a new download.

raquelalvarezbanos, 2017-01-26:
OK, let me know. I was using the latest nightly build (version 3.8.20170125.1917)

KarlPalmen, 2017-01-26:
I tried exactly the same nightly build and found out one needs to tick the checkboxes for both the periods as shown in the second illustration for test 7, for it to fail.

raquelalvarezbanos, 2017-01-26:
Not sure what's happening, it seems to work for me...

KarlPalmen, 2017-01-26:
... with the periods 1 and 2 both checked?

KarlPalmen, 2017-01-26:
It also fails on my recent build for PRs, when I made sure I ticked both periods.

raquelalvarezbanos, 2017-01-26:
Yes... with periods 1 and 2 both checked it seems to work... although fit results are not updated in the graph, but that is a different story. Below is what I have (I am afraid it is a bit small but you can open the image in a new window):

[image]

KarlPalmen, 2017-01-26:
I get this other story, if I do not run test 1 beforehand. I see `Cannot open fileError in logging framework: Cannot open file` in my cmd.exe window, but not my Mantid result log.

KarlPalmen, 2017-01-26:
I think the other story is correct behaviour, because displaying several fits together would be tricky. Individual fits are available in the workspaces created and seem OK to me.

KarlPalmen, 2017-01-26:
Given that this occurs only after changing an instrument in the course of work, which is an unlikely use case, I'm inclined to make this milestone 3.10 rather than 3.9 .

raquelalvarezbanos, 2017-01-26:
I agree. I'll move it to 3.10 and leave it unassigned. Thanks for testing the interface Karl.

closed 2021-03-02 by stale[bot]