mantidproject/mantid #18570: Indirect I(Q,t)Fit - incorrect plot options menu

louisemccann, 2017-01-26:
The Indirect I(Q, t)Fit plot options menu displays 'Beta' twice for the stretched exponential options
[image]
Intensity plots A0
Tau plots Intensity
Beta1 plots Tau
Beta2 plots Beta

### Steps to reproduce the behavior
Interfaces>Indirect>DataAnalysis>I(Q,t)Fit
Load an `_iqt` workspace
Choose Streched Exponential for fit type
Press Run
Check Plot Output menu

### Platforms affected
All

SpencerHowells, 2017-01-26:
Check the Exponential option as well. This too needs Background on the menu.

louisemccann, 2017-01-26:
@SpencerHowells that should have added Background to all options

PR #18573 (2017-02-01): Indirect I(Q,t)Fit - Update Plot Options
A new plot option has been added to I(Q,t)Fit so that Background is plotted. By doing this, the bug causing `Beta` to be listed twice and also the bug which causes the selected plot option to plot the wrong parameter should be resolved.

**To test:**
Input Files:
[irs26176_graphite002_iqt.zip](https://github.com/mantidproject/mantid/files/732557/irs26176_graphite002_iqt.zip)
Interfaces>Indirect>DataAnalysis>I(Q,t)Fit
Load workspace
Change `EndX` to be `~0.18`
Press `Run`
Check plot options menu. For `Exponential` fit types it should be:
```
Background
I [... cut]
closed 2017-02-06 by louisemccann