mantidproject/mantid #18674: Remove the old Reflectometry GUI

raquelalvarezbanos, 2017-02-02:
When scientists are happy with the new GUI, remove the old one.

gemmaguest, 2017-09-14:
There were some minor fixes required for the new GUI which will go into release 3.11. Hopefully we will be able to remove the old GUI in the next release after this.

gemmaguest, 2017-09-21:
Having spoken to Mario about SURF, I'm adding a dependency on #20088 because they currently still need to use the old interface if they need to specify different transmission runs for different angles (it is much slower to do this in the new GUI at the moment). However, they need to use the new interface if they have more than three angles.

Mario also pointed out that users may only be here once a year and may not have time to learn the new interface, so would prefer to keep the old interface around for some time.

gemmaguest, 2018-03-21:
I'm marking the dependency on #21167 complete as of v3.12 as this is done bar a couple of minor niggles. I therefore think we're ready to add the deprecation warning in v3.13

gemmaguest, 2020-02-14:
Max has confirmed that we can remove the old GUI

PR #27951 (2020-02-19): Remove old ISIS Reflectometry GUI
**Description of work.**
The old ISIS Reflectometry GUI has been removed.

**To test:**
Check that the GUI is not in MantidPlot and no warnings are shown.

Resolves #18674 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If there are changes in  [... cut]
closed 2020-02-19 by martyngigg