mantidproject/mantid #18675: Group TOF workspaces

raquelalvarezbanos, 2017-02-02:
The Reflectometry GUI produces a TOF workspace for each run. These workspaces should be grouped as the old GUI does.

PranavBahuguna, 2017-07-26:
In the `GenericDataProcessorPresenter` the `TOF` runs are being generated by a preprocessor that takes input from run numbers from the `Run(s)` column and runs the `Plus` algorithm on them, generating output workspaces with the `TOF` prefix.

Grouping these runs will not be trivial as there is currently no mechanism in the GDPP for obtaining these output runs and running a `GroupWorkspaces` algorithm on them.

gemmaguest, 2019-06-10:
I think we can do this in the algorithm `ReflectometryISISLoadAndProcess`. We can probably just hard code the group name to `TOF` and place the TOF workspaces into it when we load them. I think this is what the old GUI did.

- If the workspaces are already loaded I think we should probably group them as well - it seems that this is what the old GUI did.
- If there is already a `TOF` workspace group then we just add the additional workspaces into it.
- If there is already a workspace called `TOF` which is not a workspace group then we should prbably throw an error.
- If the input TOF workspace for a run is already a group we'll end up with something like the following where we lose the original grouping. We should check this is ok with scientists (this fails in the old GUI):
[image]

PR #25882 (2019-06-18): Group TOF workspaces
**Description of work.**
This PR groups the TOF workspaces loaded in `ReflectometryISISLoadAndPreprocess` into a group called `TOF`.

**To test:**
Run `ReflectometryISISLoadAndProcess` with the runs `INTER13461`, `INTER13462`. This should output a group called `TOF` which contains `TOF_13461`, `TOF_13462` and `TOF_13461+13462`.

Run it again with the run `INTER13463` and the workspace `TOF_13463` should be added to the `TOF` group.

Fixes #18675 .

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTick [... cut]
closed 2019-06-18 by martyngigg