mantidproject/mantid #18687: SpectrumInfo and DetectorInfo rollout: Peaks to PeaksWorkspace

ianbush, 2017-02-02:
This issue is part of the SpectrumInfo and DetectorInfo rollout, see https://github.com/mantidproject/mantid/issues/17743.
Algorithms covered in this issue are:
```
Framework/DataObjects/.../Peak.cpp
Framework/DataObjects/.../PeakTest.h
Framework/DataObjects/.../PeaksWorkspaceTest.h
```

OwenArnold, 2017-02-09:
This isn't straight forward at the moment. We should have a better idea how to do this once positions are in the `DetectorInfo`. I'm also going to have a look at what would be required to update the `InstrumentRayTracer` use the instrument view.

SimonHeybrock, 2017-02-09:
@OwenArnold Yes, we will need an equivalent of `InstrumentRayTracer`. I suspect we need a lot more basic functionality in `Beamline` before that could be done?

closed 2021-03-03 by stale[bot]