mantidproject/mantid #18714: Scan data issue

raquelalvarezbanos, 2017-02-06:
This was reported by Jos. 

Set the instrument to `OFFSPEC` and run the following script:

```
LoadISISNexus("40708",OutputWorkspace="single_period")
LoadEventNexus("40708",OutputWorkspace="single_period_t=0_onwards",FilterByTimeStart=0)
Rebin("single_period_t=0_onwards",[0,20,100000],OutputWorkspace="single_period_t=0_onwards")

LoadISISNexus("40648",OutputWorkspace="two_period")
LoadEventNexus("40648",OutputWorkspace="two_period_t=0_onwards",FilterByTimeStart=0)
Rebin("two_period_t=0_onwards",[0,20,100000],OutputWorkspace="two_period_t=0_onwards")
```

Jos expected workspaces `two_period` and `two_period_t=0_onwards` to be the same, but they are not.

raquelalvarezbanos, 2017-02-06:
I am pasting below an email I sent Jos about this issue

raquelalvarezbanos, 2017-02-06:
Hi Jos,

I was expecting that, if there were scan data from negative times, running:

test_1 = LoadEventNexus("40708", FilterByTimeStart=0)
test_2 = LoadEventNexus("40708")

would give two different workspaces, but apparently they are the same.

But going back to your problem, the issue was that, as shown in your script, when comparing the event workspace(i.e. the workspace loaded with “LoadEventNexus”) to the histogram workspace (i.e. the workspace loaded with “LoadISISNexus”), they are different. I’ve been looking into the nexus file and the loading algorithms and I think Mantid is just loading what’s in the nexus file. This is what I found when looking into this:

-	 “LoadISISNexus” essentially opens entry “raw_data_1/detector_1/counts” (see figure below). This is what gets loaded into the spectra in the loaded workspace.
-	 “LoadEventNexus” reads various things in “raw_data_1/detector_1_events/” (event_id, event_index, event_time_offset, event_time_zero, and total_counts).

[image]

-	I think Mantid is just loading what’s in the nexus file. This is how I’ve checked this. Below is the color fill plot you showed me the other day in the meeting, and as we saw there are some high intensity peaks in the `single_period` workspace:

[image]

In the figure below I am just focusing on spectrum with id 403. This spectrum has a high intensity peak at TOF ~ 97600. Specifically, bin 4876 has 13,819 counts:

[image]

This number, 13819 counts, is exactly what I find if I inspect the nexus file using HDFView (note that in the plot on the right, HDView is plotting the counts as a function of the detector ID for bin 4876):

[image]

In summary, those peaks we are seeing in the left color fill plot seem to be in the nexus file. However, when we load the run as an event run, it seems that these counts are not recorded in `detector_1_events`, I am not sure why. I am sorry I am not being more helpful, let me know if you can think of anything else I can check for you.

Raquel

gemmaguest, 2020-08-06:
Closing as stale because we are unlikely to get around to this any time soon

closed 2020-08-06 by gemmaguest