mantidproject/mantid #18730: Mantid python does not unwrap range any more

abuts, 2017-02-07:
### Currently if ones assigns range to a variable in python reduction script, it is unwrapped into list. 

### Some internal changes in mantid python, made for compliance with Python 3, changed this behaviour so if one assigns range to a variable, in fact you assigning a iterator. Direct inelastic reduction scripts fail as expect list or string as input.

Probably best way ahead in this situation (bearing in mind  future changes to Manitd) is to modify script to accept iterator.

AndreiSavici, 2017-02-07:
If you need list, just use `list(range(n))`

abuts, 2017-02-07:
I will, but number of existing scripts will break and some users will be unhappy. This is why this changes are necessary.

PR #18738 (2017-02-08): Changes to Direct reduction code to unwrap range into list and unit test for it
Fixes #18730
As I do not know how to move PR from requesting merge from master to release-3.9 this is copied from PR#18732, which have been tested by @AndreiSavici.  I have also rebased code to release-3.9 branch.

Current ISIS direct reduction scripts fail if one assigns range to it. The changes allow it to work. 

**To test:**

The unit test provided is seems good indicator to see how the changes work. I have tested reduction scripts. Test by code review. 

Users should not notice the changes. No release notes ar [... cut]
closed 2017-02-09 by martyngigg