mantidproject/mantid #18792: Indirect->Analysis, Indirect->Corrections :: Forbid selection of group workspaces.

gvardany, 2017-02-10:
The above mentioned custom interfaces crash when a group workspace is chosen. These interfaces are not supposed to operate on workspace groups, so the data selectors should explicitly forbid the group workspaces using the new feature in PR #18241.

gvardany, 2017-02-10:
@louisemccann Can you confirm that none of these tabs is supposed to work on WorkspaceGroups?

louisemccann, 2017-02-15:
@gvardany that is correct for corrections. For analysis, I think Elwin does, but it doesn't use a workspace data selector.
Could you also remember to use the Indirect Inelastic Label?

SpencerHowells, 2017-02-15:
None of our routines use a group selector. ElWin creates a Group to reduce the number of entries in the Workspace menu. If we were to use a Group in the interface what would it do? Would it run on all members of the Group (risky scientifically!) or enable selection of a member.

gvardany, 2017-02-16:
It depends what you would want, but for the moment it is not foreseen and Mantid just crashes, so it would be nice to prevent it.

SpencerHowells, 2017-02-16:
Just stopping a crash is fine by me. Go ahead.

PR #18939 (2017-03-02): Indirect>Corrections, Indirect>Analysis :: forbid group workspaces
This PR merely configures the workspace selectors in the above mentioned custom interfaces to not to accept GroupWorkspace as input. The algorithms behind are not supposed to operate over the groups. Because of mishandling, the selection of groups was causing mantid to crash as shown in the issue.

**To test:**

Pick this if you are familiar with those interfaces and have some relevant files/workspaces ready.

1. Make sure that they work as before if workspaces are given.
2. Repeat the steps in the issue, and make sure that it  [... cut]
closed 2017-03-02 by martyngigg