mantidproject/mantid #18793: HistogramData rollout: QwtWorkspaceBinData.cpp to DPDFInputDataControl.cpp

PranavBahuguna, 2017-02-10:
This issue is part of the HistogramData rollout, see #17641.
Algorithms covered in this issue are:

- MantidQt/API/src/QwtWorkspaceBinData.cpp
- MantidQt/API/src/QwtWorkspaceSpectrumData.cpp
- MantidQt/CustomDialogs/src/FitDialog.cpp
- MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/MultiDatasetFit/MDFDatasetPlotData.h
- MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCBaselineModellingView.h
- MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCDataLoadingView.h
- MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/Muon/ALCPeakFittingView.h
- MantidQt/CustomInterfaces/src/DynamicPDF/DPDFDisplayControl.cpp
- MantidQt/CustomInterfaces/src/DynamicPDF/DPDFFourierTransform.cpp
- MantidQt/CustomInterfaces/src/DynamicPDF/DPDFInputDataControl.cpp

PranavBahuguna, 2017-03-31:
Issue has been reopened as plotting has been reported to be broken after merge.

LamarMoore, 2017-03-31:
@SimonHeybrock @mdoucet the issue has been re-opened to fix the plotting issue.  When I checked his code out previously to find the source of a failing unit tests I ran all of the unit tests locally and they all passed, so this plot failing is bizarre. I am not going to be here to keep an eye on this over the next two weeks. Do you think you would be able to monitor the outcome of the fix?

SimonHeybrock, 2017-03-31:
@LamarMoore I don't think there are units tests for this, so you would not see the issue there. I will also be away for most of the next two weeks, so unfortunately I cannot have an eye on this.

PranavBahuguna, 2017-03-31:
@SimonHeybrock Do you know what plotting is being broken here?

SimonHeybrock, 2017-03-31:
See #general three days ago. The issue is the normalization flag that can be set.

martyngigg, 2019-10-22:
Closing as not worth consolidated effort in maintenance. Reveiwers can pick up and improve remaining code over time.

PR #18834 (2017-03-23): HistogramData rollout: QwtWorkspaceBinData.cpp to DPDFInputDataControl.cpp
This PR is part of refactoring effort to use the new interface of the HistogramData module.
See #17641 for details.

The following algorithms were converted to use HistogramData:

- MantidQt/API/src/QwtWorkspaceBinData.cpp
- MantidQt/API/src/QwtWorkspaceSpectrumData.cpp
- MantidQt/CustomDialogs/src/FitDialog.cpp
- MantidQt/CustomInterfaces/inc/MantidQtCustomInterfaces/MultiDatasetFit/MDFDatasetPlotData.cpp
- MantidQt/CustomInterfaces/src/DynamicPDF/DPDFFourierTransform.cpp
- MantidQt/CustomInterfaces/src/DynamicPDF/DPDF [... cut]
closed 2019-10-22 by martyngigg