mantidproject/mantid #18794: Zooming into peaks in the SplatterplotView makes rendering very slow

AntonPiccardoSelg, 2017-02-10:
When loading an event workspace into the splatter plot view the data is displayed as PointGaussian. When zooming into a peak region the VSI freezes and cannot cope with updating the rendered image in a reasonable time (takes seconds between frame updates).  When switching from PointGaussian to Points it is fine.

### Expected behavior

Frame update with little lag when zooming into a peak.

### Actual behavior

Frame updates with lags on the order of seconds when zooming into a peak.

### Steps to reproduce the behavior

1. Load a MDEventWorkspace into the splatter plot view of the VSI
2. Zoom into a peak region
  * Confirm that it gets laggy

It can be hard to zoom nicely into a peak

### Platforms affected
All

peterfpeterson, 2017-02-13:
This is not a problem for me on fc25 off of `master` with paraview v5.2.0.

AntonPiccardoSelg, 2017-02-13:
@peterfpeterson Hmm, maybe try the following:

1. Download this data set: https://www.dropbox.com/s/aesgphckbo5o1s9/ellipPeaks_new.nxs?dl=0
and this data set:  https://www.dropbox.com/s/hiux46kt71b4fty/MDEvent_ellipPeaks_new.nxs?dl=0
2. Load both data sets into Mantid. Open the VSI with the MDEvent data set.
3. Make sure you are in the splatter plot view.
4. Drag and drop the peaks workspace
5. Open Peaks->Show all peaks in table
  * You should get the peaks table Click on one and it should zoom into a peak and set the rotation    point into the centre of the peak.
6. Now zoom futher into the peaks until the Gaussian peaks become large
  * Confirm that the rendering becomes laggy.

Let me know if this is a problem on your machine

peterfpeterson, 2017-02-13:
That is a much better description of the issue. I'm not sure what is happening, but that somehow takes over my mouse and all I can do is shutdown mantid to get my system back.

AntonPiccardoSelg, 2017-02-14:
I am not sure if this is the reason, but we are essentially performing volume rendering on a large part of the screen. The low opacity means we have to follow the ray over many layers.  Turning the opacity up to 1 or using the Points allows for z culling which would be more efficient. No idea if this is completely right, but it seems plausible to me. I am just surprised that my NVidia card cannot handle this. 

I don't have to switch off my computer to get back control, I  just have to zoom out by turning the mouse wheel -- it  takes a while though(~1/2 minute).

quantumsteve, 2017-02-14:
@AntonPiccardoSelg Which platform are you testing on? I do see some lag on MacOS, but it's nothing like the 30 second or longer delay you and @peterfpeterson are experiencing.

quantumsteve, 2017-02-14:
I did profile zooming in and out of a peak and confirmed 100% GPU usage. Specifically, it looks like there are several synchronous, blocking OpenGL calls.

https://github.com/Kitware/VTK/blob/ef07338e3ded86dd6600c4576de61085e0687329/Rendering/OpenGL2/vtkDualDepthPeelingPass.cxx#L903

Can we make this call asynchronous?
https://vertostudio.com/gamedev/?p=177
https://books.google.com/books?id=Nwo0CgAAQBAJ&lpg=PT763&dq=glGetQueryObjectuiv%20slow&pg=PT763#v=onepage&q=glGetQueryObjectuiv%20slow&f=false

https://github.com/Kitware/VTK/blob/ef07338e3ded86dd6600c4576de61085e0687329/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx#L1039

Would a PBO speed this up?
https://vec.io/posts/faster-alternatives-to-glreadpixels-and-glteximage2d-in-opengl-es

AntonPiccardoSelg, 2017-02-15:
@quantumsteve  I see a lag on an Ubuntu 16.04 Desktop with a Nvidia graphics card. I also saw the same issue on a Windows 7 Laptop (Intel HD integrated graphics card) and finally on Owen's Mac with an Nvidia card, but there it wasn't as bad as I saw on me machines.

What did you use to profile the GPU usage?

quantumsteve, 2017-02-15:
Apple Instruments [GPU Driver](https://developer.apple.com/library/content/documentation/DeveloperTools/Conceptual/InstrumentsUserGuide/Instrument-GPUDriver.html) and time profiler.

quantumsteve, 2017-09-08:
This appears to have been addressed for ParaView v5.5

https://blog.kitware.com/major-improvements-on-the-point-gaussian-representation/

quantumsteve, 2017-09-11:
@AntonPiccardoSelg Please open a new issue if the above improvements are insufficient. In that case, we'd likely have to bring in someone fluent in the OpenGL API.

closed 2017-09-11 by quantumsteve

AntonPiccardoSelg, 2017-09-22:
@quantumsteve I am just performing unscripted testing on Vates and it seems that this issue is still here. My computer and graphics card are old however, I am wondering if this might be fixed for other people (with relatively up-to-date hardware). Did you test the behaviour on your machine? If not, would you mind testing that? Thanks.

quantumsteve, 2017-09-22:
I can't reproduce the issue locally, but I also have a new computer with a large GPU. The profiling above suggests this does depend a lot on the user's system. 

Transparency was set to allow one to see inside of each peak, not just the first layer. Without user complaints, I'm hesitant to return to opaque spheres or points. Similarly, I don't have time to study asynchronous OpenGL functions and/or rewrite the code used for coloring vertices so one doesn't need to look inside.