mantidproject/mantid #18820: A failed algorithm can't be canceled

raquelalvarezbanos, 2017-02-13:
### Expected behavior

Mantid should be able to cancel an algorithm that failed and it shouldn't hang if you try to close MantidPlot.

### Actual behavior

A failed algorithm can not be canceled. Then, MantidPlot hangs if you try to close it.

### Steps to reproduce the behavior

- Load `POLREF14882.nxs` which is a multi-period workspace
- Open `ConvertUnits` from the algorithm dialog. Select `POLREF14882` and set `Target=MomentumTransfer` and `AlignBins=True`.
- Try to cancel the algorithm from `Details`.

### Platforms affected

All?

gemmaguest, 2017-02-13:
This is not a regression in v3.9.0, as I had thought - I can reproduce similar behaviour in v3.8.0, although with a different test case (because `ConvertUnits` doesn't throw in v3.8.0):

```
inWS=Load(Filename=r'POLREF00014882.NXS')
IvsQ, IvsQ_unbinned, IvsLam = ReflectometryReductionOneAuto(InputWorkspace=inWS)
```

The problem happens using both the GUI or a script. Confirmed on Windows 7 and Ubuntu 16.04.

gemmaguest, 2017-02-13:
The problem is that an `ErrorNotification` does not get sent for the parent algorithm. One is sent for the first child algorithm, but the child is not registered with the `AlgorithmMonitor`, so no notification is received (i.e. here: https://github.com/mantidproject/mantid/blob/master/MantidPlot/src/Mantid/AlgorithmMonitor.cpp#L103)

I think the solution is to catch the error and send an `ErrorNotification` for the parent algorithm here: https://github.com/mantidproject/mantid/blob/master/Framework/API/src/Algorithm.cpp#L531

I think this should be straightforward. I just need to check whether any other error handling needs to be done at this point.

gemmaguest, 2017-02-16:
After talking to @martyngigg about this, we decided the best place for the `ErrorNotification` would be inside `doCallProcessGroups`, because this is where the `FinishedNotification` is: https://github.com/mantidproject/mantid/blob/master/Framework/API/src/Algorithm.cpp#L1194

PR #18943 (2017-03-02): Send error notifications for parent algorithms
This PR fixes a problem where error notifications are not sent from a parent algorithm if one of its child algorithms throws an exception. This causes a problem where the algorithm monitor still thinks the algorithm is running and it cannot be cancelled from the Details dialog. MantidPlot also hangs when you try to close it because it is waiting for the algorithm to finish.

This PR also adds some additional tests to check that the correct notifications are received when running algorithms on workspace groups.

**To test:**

Run the following scri [... cut]
closed 2017-03-02 by peterfpeterson