mantidproject/mantid #18821: Make it possible to select different corrections based on selected monitors

raquelalvarezbanos, 2017-02-13:
Scientists would like to be able to select different polynomial/exponential corrections depending on the monitor of interest. Currently there's only one set of corrections defined in the parameter file.

This issue is to discuss with them how this should be implemented.

gemmaguest, 2017-03-27:
This should be easy so may make it into 3.10 if we have time. Christy can explain what needs to be done.

gemmaguest, 2018-06-18:
This has been pushed back to v3.14 due to other work taking priority

gemmaguest, 2019-07-23:
I'm removing the milestone until we review priorities because it turns out this is more complicated than initially envisaged.

gemmaguest, 2020-08-06:
Closing as Stale because we are unlikely to get around to this any time soon. Most of what we need is already there and the final parts sound quite complex and low priority.

closed 2020-08-06 by gemmaguest