mantidproject/mantid #18825: Buttons to expand/collapse all groups

raquelalvarezbanos, 2017-02-13:
Scientists mentioned they would like to additional buttons in the Reflectometry interface to be able to expand/collaps all the groups in the table.

gemmaguest, 2017-03-02:
This is low priority but may make it into 3.10 as it is an easy win

raquelalvarezbanos, 2017-03-02:
Just for information: the class that is creating the "actions" (also called "commands" in the code) is `DataProcessorTwoLevelTreeManager`. Two new actions will have to be created and class `ReflRunsTabPresenter` will have to be modified accordingly (see `ReflRunsTabPresenter::pushCommands()`, not sure if other classes may need to be modified too). The new actions will take a reference to the `GenericDataProcessorPresenter` (see for instance [AppendRowCommand](https://github.com/mantidproject/mantid/blob/master/MantidQt/MantidWidgets/inc/MantidQtMantidWidgets/DataProcessorUI/DataProcessorAppendRowCommand.h)) which can then act on the view with methods `QTreeView::expandAll()` and `QTreeView::collapseAll()`.

raquelalvarezbanos, 2017-03-02:
If you ever have any questions/doubts about the `DataProcessorUI` widget the person to ask is Owen.

PranavBahuguna, 2017-03-08:
Should there be two separate buttons for expanding and collapsing all groups respectively, or should both be handled by the same button?

PR #19241 (2017-03-29): ISIS Reflectometry GUI: Added buttons to expand/collapse all groups
This PR adds two buttons to the interface on the `Runs` tab:
- Expand groups - expands all currently closed groups in the table.
- Close groups - closes all currently open groups in the table.

**To test:**
- Code review.
- The interface can be accessed from Interfaces->Reflectometry->ISIS Reflectometry.
- Add several groups to the table. Test that each of the new buttons opens and closes all groups respectively.
- Check that documentation has been updated accordingly.

Fixes #18825.

*Release notes have been updated.*

#### R [... cut]
closed 2017-03-29 by peterfpeterson