mantidproject/mantid #18826: Reflectometry GUI: save to file

raquelalvarezbanos, 2017-02-13:
This issue is to make it possible to save/load all the information in the GUI to a file.

gemmaguest, 2017-03-27:
Currently it is possible to save the table, but users want to be able to save all of the information from all tabs in the GUI, including the table, into a single file.

gemmaguest, 2017-09-07:
We should look at this together with SANS, who would like something similar - they currently load settings from user files, but can't save state from the GUI. They also want a better format, e.g. json or yaml. There are some notes on this in the [Users Workshop minutes](https://github.com/mantidproject/documents/blob/master/Presentations/DevMeetings/2017-06/SANS_NR_minutes_breakout_2017.md) (see the the **User files** section)

edwardb96, 2018-03-28:
Discussion with @Mantid-Matthew has suggested that SANS are keen to use the same file format as Reflectometry. Saving to a file from the SANS GUI is not however a high priority item. So it is likely that Reflectometry will be the first to implement this. The SANS scientists would like to be kept in the loop and involved in discussions over which format is used.

Matthew has identified the following additional requirements from the SANS scientists which may or may not be requirements for reflectometry as well:
- Ability to open the file in a simple text editor rather than something like hdfview.
- Ability to comment out and change sections of the file.

Bearing these in mind we considered the following formats to start.
## XML
```xml
<?xml version="1.0" encoding="UTF-8" standalone="yes" ?>
<root>
  <foo>bar</foo>
  <numbers><!-- A comment -->
    <number>1</number>
    <number>2</number>
    <number>3</number>
  </numbers>
  <subobject some_value="true">
    <subsubobject some_other_value="false"></subsubobject>
  </subobject>
</root>
```
- Hardest to read
- `` to close.
- Largest file size.
- Libraries available in Mantid on C++ and Python side.
 
## JSON
```json
{
  "foo" : "bar",
  "numbers": [ 1, 2, 3 ],
  "subobject": {
     "some_value": true,
     "subsubobject": {
       "other_value": false
     }
  }
}
```
- Heavily Punctuated 
- Cannot comment things out.
- Libraries available in Mantid on C++ and Python side.

## YAML
```yaml
---
foo: "bar"
numbers: # A comment
  - 1
  - 2
  - 3
subobject:
  some_value: true
  subsubobject:
    other_value: false
...
```
- Less punctuated than JSON or XML.
- Whitespace denotes strucure - good for readability, not so much for reducing user errors.
- Some complex/obscure syntax in places.
- Libraries available in Python but more limited options for C++ which is a problem for reflectometry.

## TOML
```toml
foo = "bar"
numbers = [ 1, 2, 3 ] # A comment

[subobject]
some_value = true

  [subobject.subsubobject]
  other-value = false
```

- Extended + formally specified version of INI.
- Comments using `#`
- Used by Cargo, Rust's package manager.
- Less mature than others on this list.
- Packages available for Python and Libraries (Incl two modern header only libraries) available for C++.

Based on the above, my personal recommendation would be to use TOML, subject to further investigation and discussion with the instrument scientists.

As an additional note, it is possible that when implementing this we could consider adding a solution to the feature:

> Jos would like to be able to load them from a nexus file which could be created on ICat or local directory
> in mantid's path on the fly and found by the GUI just by specifying its name. Ideally it would then also 
> populate the text box which could then be edited by hand.

from #21167.

smk78, 2018-03-29:
A little historical perspective from the SANS side...

So the way we reduce data in Mantid from a GUI essentially emulates how we used to do things in our previous (VMS based) data reduction programe 'COLETTE' (the name being a corruption of COrrect Laboratoire d'Electronique des Technologies de l'Information, which would have been COLETI; COLETTE just seemed nicer and more memorable. We'd probably get into trouble for it these days!!! LETI made the first LOQ detector). COLETTE actually sat on top of VMS GENIE (the forerunner to OpenGenie used in SECI), which it used for data i/o and plotting. But because GENIE was written by the crystallographers/spectroscopists it knew nothing about 2D detectors, so we (principally Ray Osborn, Richard Heenan & myself) had to create new routines and command sets to augment GENIE to reduce LOQ data. At the same time that gave us the flexibility to employ the VMS screen management library to create what, today, people would recognise as a GUI. But the only human input device back then was a keyboard, so we needed an efficient way of setting up all the reduction parameters. That became a file of COLETTE/GENIE commands we called a 'Mask File'. And it worked very well. So when Mantid came along we had the same concept implemented. The only change has been that for technical reasons we now call the Mask File a 'User File'. But the bottom-line is this is a concept that has been around 30+ years and which our user community understand and are comfortable with.

Essentially the User File is a SANS reduction configuration file for Mantid. I'll attach a LOQ example and a SANS2D example. As you'll see, there are many similarities, but also many differences. And now we have LARMOR and ZOOM too. So the time has definitely come to impart some structure on these files! But they absolutely, most definitely, have to be human-readable, and editable in nothing more sophisticated than Notepad (if really needed; I take the point that a language-sensitive editor like, say, Notepad++ might convey many advantages).

The User Files are generally created on an experiment-by-experiment basis (usually by copying the last one, renaming it, and making some subtle changes), and any single experiment may have several, depending on the sophistication of the reduction. We give the User Files to the users with their data. Equally, many users will copy/edit their own versions. So LOADING these files into the GUI is critical.

Creating a User File by SAVING the state of the GUI is not something which, by its absence, we have ever found to be a problem (after all, how hard is it to copy a file, rename it, and edit it!), but it would be slightly less laborious than the current need to keep reloading an edited User File when you are testing changes. So we would welcome such a feature. It's just not a high priority for SANS.

[USER_LOQ_174M_M4_Wang_12mm_Changer_MAIN.txt](https://github.com/mantidproject/mantid/files/1859774/USER_LOQ_174M_M4_Wang_12mm_Changer_MAIN.txt)

[USER_SANS2D_174K_2p4_4m_M3_Barlow_8mm_Changer.txt](https://github.com/mantidproject/mantid/files/1859777/USER_SANS2D_174K_2p4_4m_M3_Barlow_8mm_Changer.txt)

smk78, 2018-03-29:
In terms of a new format for User Files, I had been wondering about something not unlike a a Windows .INI file. But maybe that is too basic?

We use XML for one of our output data formats, and it presents quite nicely, but it is 'wordy'.

I have come across JSON and am not a big fan. I don't find the syntax/structure very user-friendly.

YAML I've heard of but have no experience. TOML I've never heard of!

Does someone want to take one of the User Files I just uploaded and TOML-ise it as a discussion example?

edwardb96, 2018-03-29:
With some help from @Mantid-Matthew and the [documentation](https://www.mantidproject.org/SANS_User_File_Commands) I have put this together... Let me know your thoughts.
```toml
[instrument]
name = "LOQ"

[detector]
type = "Main"

[gravity]
enabled = true
extra_length = 2.0

[limits]
radial = {min = 35, max = 419}
wavelength = {min = 2.2, max = 10.0, step = 0.035, type = "log"}

  [limits.detectors]
  main = {min = 0.008, max = 0.255, step = 0.002, type = "lin", unit="Q"}
  #main = {min = 0.029, max = 0.149, step = 0.08, type="log", unit="Q"}
  #main = {min = 0.007, max = 0.255, step = 0.04, type="log", unit="Q"}
  #main = {min = 0.007, max = 1.4, step = 0.08, type="log", unit="Q"}

  hab = {min = 0.22, max = 0.8, step = 0.0125, type = "lin", unit="Q"}

  [limits.detector.merged]
  fit = true
  range = {min = 0.12, max=0.24}

  [limits.qxy]
  range = {min = 0, max = 0.1, step = 0.001, type = "lin"}

[monitors]
spectrum_number = 2
transmission_spectrum_number = 2
direct_file = 'DIRECT_LOQ_MAIN_143B.txt'
hab_file = 'DIRECT_LOQ_HAB_983.txt'

  [monitors.background_count]
  times = [31000, 39000]
  # Use tof_window = {spectrum_number = 2} with beamstop M4 to avoid zero transmissions at longest wavelengths
  tof_window = {spectrum_number = 2, times = [31000, 39000]}

  [monitors.flat]
  rear_file = 'FLAT_CELL_LOQ_MAIN_143.txt'
  #front_file = 'FLAT_CELL_LOQ_HAB_BanksAvgdSeparately_131.txt'

# Distance from front surface of Ordela to middle of scintillator is ~12mm
[transmission]
spectrum_number = 17788
#spectrum_number=3
shift = -12
# Using beamstop M4 for transmissions

  [transmission.fit]
  function = "Polynomial"
  coefficients = [3, 2.2, 10.0]

[qresolution]
enabled = true
deltar = 3.0
source_aperture = 20.0
sample_aperture = 12.0
collimation_length = 4.0
moderator = 'ModeratorStdDev_TS1_Hydrogen_30Jul2015.txt'

[mask]
spectrum_numbers = [
  1042,
  1050,
  1058,
  1083,
  1090,
  1106,
  1114,
  1122,
  1138,
  1146,
  1150,
  1595,
  1658,
  6266,
  10362,
  14458,
  14460,
  14462,
  14464,
]

#spectrum_number_ranges = [
#  {begin=116, end=126}
#]

columns = [
  0,
  127
]

rows = [
  0,
  127
]

tofs = [
  {begin = 19711.5, end = 21228.5},
  {begin = 39354.5, end = 41348.5}
]

lines = [
  {width = 9, angle = 151.0}
]

maskfile = 'LOQ-above-beamstop-crescent.xml'

# Set offset value = (-11.0 + Lsm) x 1000
[sample]
offset = 2.0

[centre.main]
x = 323.165
y = 329.177
d1 = 5.00
d2 = 5.00

[centre.hab]
x = 326.040
y = 330.927

[scales]
s = 1.4901
a = 1.179
b = 1.233
c = 1.244
d = 1.216

[metadata]
mantid_interface = "SANS v2"
format_version = "1.0.0"
```

smk78, 2018-04-03:
So I see what you have done here, structuring by user file command. And as a first pass, it is certainly an option. But it is not actually the cleanest structure, because "functionally-similar" items could be grouped together more. For example, the direct_ files and the flat_ files are all detector response correction files (and in the future we might also have dark current and deadtime files too) but are currently separated in odd places because of the command keyword that invokes them. Whereas I think it would make more sense if they were under something like [detector.corrections], or maybe even just [corrections]. And [limits] is a real hotchpotch because it is such a well-used command!

How feasible would it be to address this do you think?

smk78, 2018-04-03:
It might be easier to sit down with a printout...

edwardb96, 2018-04-18:
Here is a very quick attempt at doing something similar for reflectometry... https://gist.github.com/edwardb96/e876a2e76362153c7dd458b34bdd741d

edwardb96, 2018-06-21:
> I’ve taken that initial TOML-ised user file that @edwardb96 / @Mantid-Matthew produced and heavily edited it into a structure > that I think would work for me and SANS2D.
>
> But I realise that doesn’t necessarily mean it would work for LARMOR/ZOOM or NR, or that it is practical from a Mantid perspective. But I’m throwing it out to you now so you can see my thinking and some of the questions I have.
> @smk78 
[TOML-User-File-v2.txt](https://github.com/mantidproject/mantid/files/2123729/TOML-User-File-v2.txt)

gemmaguest, 2019-08-29:
Re-opened as a SANS issue as this contains relevant information. The SANS requirement is mainly to replace the existing file format which is used for setting the GUI state from the file. We should check if saving state from the GUI is required (issue #3058 originally requested to do that but I believe it is a low priority compared to loading).

smk78, 2019-09-02:
In the case of ISIS SANS much of the information populating the GUI is, of course, loaded in from the User File, which NR do not (yet) have. So, yes, moving the User File format to TOML is the greater priority for SANS.
The question that then arises is, what other information does saving the GUI state record that is not in our User Files? Run numbers in the reduction table are an obvious one, but these would be captured by the proposed changes to Project Save/Load. Is there anything else of particular note?

gemmaguest, 2019-09-04:
I've reworded my previous comment as it's about save/load of a User File rather than project save/load. Here's an attempt to summarise the differences, and where we're up to:

**Project save/load/recovery**
This is concerned with restoring the full state of the GUI, including anything in the table, the rows' processed states, any outputs e.g. from the beam centre finder, etc. etc. It would be for backup/recovery so would not be intended to be user-readable. This is implemented for ISIS Reflectometry and was relatively straightforward to just scrape everything directly off the GUI.

**User File save/load**
The User File is more concerned with the configuration settings for a particular reduction. Swapping out the parser used so that we can load from a different file format such as TOML should hopefully be relatively straightforward.

Saving is less straightforward due to the requirement to preserve user-friendly formatting and comments. I don't think in the existing SANS workflow we would expect to create a new User File from scratch using the GUI? It sounds like saving might be most useful as a way to update an existing User File to avoid having to re-load it after making changes. If that's the case, maybe a simple Refresh button would be ok, or watching the file for changes so that it automatically updates?

This differs from ISIS Reflectometry workflow where the GUI is the starting point. We have implemented functionality to save the entered information to a "batch" file which contains all of the runs and settings for a specific reduction. This a bit like the SANS User File and Batch file combined, but is not intended to be easily user-readable - we will effectively use the GUI like an editor to load and update it. So, we sort of do have a User File, but it's used in a different way.

DavidFair, 2020-02-18:
The majority of this implementation is done, however mapping the TOML syntax to Sans State objects is revealing some unimplemented features / assumptions I'm having to run past

This has lead to concerns from myself and discussed with the instrument scientists about how we might end up releasing something "half-baked". So I'm pushing this off for 5.1 to go into the nightly fairly early on in development

gemmaguest, 2020-09-07:
Closing because the new user file is now implemented. There are a couple of issues left in this epic but they are more follow on tasks so I don't think we need the epic any more.

PR #26625 (2019-08-21): ISIS Reflectometry: Save and Load individual batches
**Description of work.**
This will allow users to Save and Load individual batches for the ISIS Reflectometry GUI.

**To test:**
- Open Workbench
- Open ISIS Reflectometry
- Do some things and change some settings
- Click Batch->Save and save it somewhere
- Close the ISIS Reflectometry GUI
- Open it again and click Batch->Load
- All settings should be back and the table should be in a close but not completely the same state, which is unprocessed.

- Repeat for MantidPlot as they have different backbones for file writing.

Fixes #18826. 

--- [... cut]
closed 2020-09-07 by gemmaguest