mantidproject/mantid #18829: New option to `ReflectometryReductionOneAuto` to sum in Q

raquelalvarezbanos, 2017-02-13:
This issue is to add a new input property to `ReflectometryReductionOneAuto` to sum in `Q`. Currently the algorithm sums in wavelength when running `GroupDetectors`. It should be possible to sum in `Q` as well.

raquelalvarezbanos, 2017-03-06:
As Jos mentioned, to calculate the contribution of a pixel to a "virtual" pixel we need to take into account the overlap between them. In Mantid, we have something called `FractionalRebinning` (see [here](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/src/FractionalRebinning.cpp#L115-L165)) which is used in `ConvertToReflectometryQ` (see `ReflectometryTransform::executeNormPoly()`). I am including below a sketch of how the algorithms works. The red square is in this case our input pixel with intensity I0 (in red), and let's assume area A=1. `FractionalRebinning` calculates the contribution of this red pixel to each of the "virtual" blue pixels by normalizing the intensity in the red pixel, I0 (in red), by the overlap region.

[image]

I think the idea is similar to what we need to do, but there are some differences that may make this `FractionalRebinning` unusable in our case.

In fact, thinking about it, I am not sure we can reuse it, so just keep in mind that this functionality is in Mantid, in case you think it could be useful.

gemmaguest, 2017-03-10:
Here are the rough steps I think need to happen. This will replace the summing in lambda currently done by `ReflectometryWorkflowBase2::makeDetectorWS()`. The output `IvsLam` will be a 1D array of counts vs "virtual lambda".

**Inputs**
- Workspace
- Processing instructions and wavelength range (specifies area of interest)
- Whether divergent beam or non-flat sample (cannot be both)

**Normalise**
- Normalise the 2D input in `(twoTheta, lambda)` by monitor and transmission run

**Find constants**
- `thetaR` = midpoint of `thetaMin` and `thetaMax` (from area of interest)
- *divergent beam:* `theta0` = the theta value of the centre pixel e.g. pixel 404 for OFFSPEC
- *non-flat sample:* `theta0` = 0

**Construct the output array `IvsLam`**
- Find the extents of output array in virtual lambda using eq5:
  `lambdaVMax` is the projected `lambdaTop` from pixel `[lambdaMax, twoThetaMin]`
  `lambdaVMin` is the projected `lambdaBot` from pixel `[lambdaMin, twoThetaMax]`
- NB We may not need this full range because the output will be cropped (needs further investigation).
- Use the same bin widths as the input?

**For each input value at `[twoThetaR + gamma, lambda]`**
- Find the projected range in virtual lambda using eq5:

  `lambdaTop = sin(twoThetaR - theta0)(lambda + bLambda/2) / sin(twoThetaR - theta0 + gamma - bTwoTheta/2)`
  `lambdaBot = sin(twoThetaR - theta0)(lambda - bLambda/2) / sin(twoThetaR - theta0 + gamma + bTwoTheta/2)`

  where `bTwoTheta` is the range of theta for the current detector pixel and `bLambda` is the input bin width.

- Share counts from the input pixel between the output bins that overlap the range `(lambdaBot, lambdaTop)`. Counts should be shared proportionally depending on how much of the output bin range `(lambdaStart, lambdaEnd)` overlaps the projected range `(lambdaBot, lambdaTop)`, i.e.

  `bin_counts += (input_counts * overlap_width) / (lambdaTop - lambdaBot)`

  where  `overlap_width = min(bLambda, lambdaTop - lambdaStart, lambdaEnd - lambdaBot)`

- NB Again, we may not need to process every input pixel because the output will be cropped (needs further investigation)

raquelalvarezbanos, 2017-03-23:
I think the steps you describe are correct. I can't spot anything else that we would need to do at this point (except for the possible future improvements you describe).

PR #19502 (2017-05-22): Add option to `ReflectometryReductionOne` to sum in Q
This PR adds an option, `SummationType`, to `ReflectometryReductionOne` to allow summation in Q, as well as in the default lambda. The option is also available in `ReflectometryReductionOneAuto` and via the Options field in the reflectometry GUI. In addition, there is an option, `ReductionType`, to specify what type of reduction should be done when summing in Q (`DivergentBeam` or `NonFlatSample`).

Documentation will follow in a separate PR.

**To test:**

Interim test scripts are below, but the results still require scientific validation. [... cut]
closed 2017-05-22 by martyngigg