mantidproject/mantid #18840: LoadFits Crashes MantidPlot

OwenArnold, 2017-02-14:
Running `LoadFits` with `LoadAsRectImg` set True causes Mantid to crash as follows.

The fits file I am trying to load is located here. I saved it using `SaveFits`. There are 5 bins and 262144 histograms in the workspace I save. The saved fits file is located here:

https://drive.google.com/open?id=0Bx07Hfz4WgYMQWZjRFNRYi1UUm8

This is the crash stack trace.
```
Thread 3 Crashed:
0   libMantidDataHandling.dylib   	0x000000011ba28d00 Mantid::DataHandling::LoadFITS::readDataToWorkspace(Mantid::DataHandling::FITSInfo const&, double, boost::shared_ptr<Mantid::DataObjects::Workspace2D>, std::__1::vector<char, std::__1::allocator<char> >&) + 832
1   libMantidDataHandling.dylib   	0x000000011ba27d62 Mantid::DataHandling::LoadFITS::makeWorkspace(Mantid::DataHandling::FITSInfo const&, unsigned long&, std::__1::vector<char, std::__1::allocator<char> >&, std::__1::vector<std::__1::vector<double, std::__1::allocator<double> >, std::__1::allocator<std::__1::vector<double, std::__1::allocator<double> > > >&, std::__1::vector<std::__1::vector<double, std::__1::allocator<double> >, std::__1::allocator<std::__1::vector<double, std::__1::allocator<double> > > >&, boost::shared_ptr<Mantid::DataObjects::Workspace2D>, bool, int, double) + 1202
2   libMantidDataHandling.dylib   	0x000000011ba23565 Mantid::DataHandling::LoadFITS::doLoadFiles(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool, int, double) + 1365
3   libMantidDataHandling.dylib   	0x000000011ba2230d Mantid::DataHandling::LoadFITS::exec() + 589
4   libMantidAPI.dylib            	0x00000001060d4f98 Mantid::API::Algorithm::execute() + 1672
5   libMantidAPI.dylib            	0x00000001060dc19c Mantid::API::Algorithm::executeAsyncImpl(Poco::Void const&) + 28
6   libMantidAPI.dylib            	0x00000001060f0363 Mantid::API::AlgorithmProxy::executeAsyncImpl(Poco::Void const&) + 35
7   libMantidAPI.dylib            	0x00000001060f1e70 Poco::ActiveRunnable<bool, Poco::Void, Mantid::API::AlgorithmProxy>::run() + 64
8   libPocoFoundation.dylib       	0x0000000106e02e82 Poco::PooledThread::run() + 108
9   libPocoFoundation.dylib       	0x0000000106e00ff0 Poco::ThreadImpl::runnableEntry(void*) + 106
10  libsystem_pthread.dylib       	0x00007fff987e005a _pthread_body + 131
11  libsystem_pthread.dylib       	0x00007fff987dffd7 _pthread_start + 176
12  libsystem_pthread.dylib       	0x00007fff987dd3ed thread_start + 13
```

DimitarTasev, 2017-02-14:
I'll have a look, there are several other known issues with LoadFits #18242, but not much can be done with the current workspace structure.

OwenArnold, 2017-02-14:
It definitely shouldn't crash MantidPlot. I appreciate that some additional information might need to be provided by the user to complete the loading/saving because it won't be available in the native file format.

DimitarTasev, 2017-03-30:
I am not sure why the image you saved out crashes Mantid, the cause is that a `std::reverse_copy` has an out of bounds error so it segfaults. I tried loading the image you saved out with ImageJ and nothing can be seen in the image window, but it doesn't crash.
You should also be aware that the data from the workspace you saved was trampled down to int16 range, so all of the values are now 0, 1 and 2.
I could not replicate the issue with an actual image and the workflow `LoadFITS > SaveFITS > LoadFITS on saved image`.
Possible cause would be 
```
const size_t start = ((i * (bytespp)) * nrows) + (j * (bytespp));
uint8_t const *const buffer8Start = buffer8 + start;
// Reverse byte order of current value. Make sure we allocate enough
// enough space to hold the size
uint8_t byteValue[g_maxBytesPP];
std::reverse_copy(buffer8Start, buffer8Start + bytespp, byteValue);
```
doesn't like not-rectangle shaped internal data structure (i.e. 512x512 image is fine, but 5x262444 will crash). 
That is something that should be fixed only if any further work is done on MDImageWorkspace. Until then this is low priority.

closed 2019-07-25 by NickDraper