mantidproject/mantid #18861: ParameterMap of const workspaces such as input workspaces can be modified.

SimonHeybrock, 2017-02-15:
`Instrument::getParameterMap()` is `const` and returns a `shared_ptr` to its internal `ParameterMap`:
https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument.cpp#L138

As a consequence it is possible to modify the map for const workspaces such as workspaces obtained from an input property. Modifying the map allows for changing positions and rotations of detectors, amongs others.

Maybe the biggest issue this causes is related to locking of workspaces: `Algorithm` locks input workspaces for reading. **There can be multiple readers.** Thus, there might be crashes, or worse, unpredictable results of another algorithm being run at the same time.

SimonHeybrock, 2017-02-15:
I think this is actually even worse:

- `ExperimentInfo::setInstrument` will *shared* the `ParameterMap`.
- Amongst others, `WorkspaceFactory::create(parent)` uses `child.setInstrument(parent.getInstrument())`, sharing the `ParameterMap`.
- `child.getInstrument()->getParameterMap()` can now be modified and will (I think) affect also `parent`.

Note that there is a copy-on-write mechanism for the `ParameterMap` in `ExperimentInfo::instrumentParameters()` so any code that obtains the `ParameterMap` via this is ok.

martyngigg, 2017-02-15:
@SimonHeybrock Do we know what level of usage we are talking about with  `Instrument::getParameterMap`?

I have mostly seen the parameter map retrieved via `ExperimentInfo::instrumentParameters()`.

SimonHeybrock, 2017-02-15:
@martyngigg Yes, it is not used a lot:

```
$ grep -r getParameterMap
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "I0MonitorIndex", 1.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "PointDetectorStart", 1.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "PointDetectorStop", 1.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "LambdaMin", 0.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "LambdaMax", 10.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "MonitorBackgroundMin", 0.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "MonitorBackgroundMax",
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "MonitorIntegralMin", 0.0);
MantidQt/MantidWidgets/test/DataProcessorUI/GenericDataProcessorPresenterTest.h:    inst->getParameterMap()->addDouble(inst.get(), "MonitorIntegralMax", 10.0);
Framework/Geometry/test/ComponentHelperTest.h:    auto pmap = inst->getParameterMap();
Framework/Geometry/test/ComponentHelperTest.h:    auto pmap = inst->getParameterMap();
Framework/Geometry/test/ComponentHelperTest.h:    auto pmap = inst->getParameterMap();
Framework/Geometry/test/ComponentHelperTest.h:    auto pmap = inst->getParameterMap();
Framework/Geometry/test/InstrumentTest.h:    TS_ASSERT_THROWS(i.getParameterMap(), std::runtime_error);
Framework/Geometry/test/InstrumentTest.h:    TS_ASSERT_THROWS(i.getParameterMap(), std::runtime_error);
Framework/Geometry/inc/MantidGeometry/Instrument.h:  boost::shared_ptr<ParameterMap> getParameterMap() const;
Framework/Geometry/src/Instrument.cpp:ParameterMap_sptr Instrument::getParameterMap() const {
Framework/Geometry/src/Instrument.cpp:    throw std::runtime_error("Instrument::getParameterMap() called for a "
Framework/Geometry/src/Instrument.cpp:  auto pmap = boost::make_shared<ParameterMap>(*getParameterMap());
Framework/Crystal/test/OptimizeCrystalPlacementTest.h:    auto pmap = inst->getParameterMap();
Framework/Crystal/src/OptimizeCrystalPlacement.cpp:      OldInstrument->getParameterMap();
Framework/Crystal/src/LoadIsawPeaks.cpp:  ParameterMap_sptr parMap = instr->getParameterMap();
Framework/Crystal/src/PeakHKLErrors.cpp:      instSave->getParameterMap();
Framework/Crystal/src/SCDCalibratePanels.cpp:      instrument->getParameterMap();
Framework/Crystal/src/SCDCalibratePanels.cpp:  boost::shared_ptr<ParameterMap> pmap = NewInstrument->getParameterMap();
Framework/Crystal/src/SCDCalibratePanels.cpp:  boost::shared_ptr<ParameterMap> pmap = NewInstrument->getParameterMap();
Framework/Crystal/src/SCDCalibratePanels.cpp:  ParameterMap_sptr pmap = instrument.getParameterMap();
Framework/Algorithms/test/ReflectometryReductionOneAutoTest.h:    inst->getParameterMap()->addDouble(inst.get(), "I0MonitorIndex", 1.0);
Framework/Algorithms/test/ReflectometryReductionOneAutoTest.h:    ParameterMap_sptr params = tinyInst->getParameterMap();
Framework/Algorithms/test/ReflectometryReductionOneAutoTest.h:    ParameterMap_sptr params = tinyInst->getParameterMap();
Framework/Algorithms/test/FindPeaksTest.h:    getParameterMap(outtablews, 0, parammap);
Framework/Algorithms/test/FindPeaksTest.h:  void getParameterMap(TableWorkspace_sptr tablews, size_t rowindex,
Framework/Algorithms/src/ClearInstrumentParameters.cpp:  const ParameterMap_sptr params = instrument->getParameterMap();
Framework/API/src/ExperimentInfo.cpp:    const auto pmap = oldInstr.getParameterMap();
Framework/API/src/ExperimentInfo.cpp:    m_parmap = boost::make_shared<ParameterMap>(*instr->getParameterMap());
Framework/DataHandling/test/LoadISISNexusTest.h:    auto params = inst->getParameterMap();
Framework/DataHandling/test/LoadEventNexusTest.h:    auto params = inst->getParameterMap();
Framework/DataHandling/test/LoadEventNexusTest.h:    auto params = inst->getParameterMap();
Framework/DataHandling/src/LoadIDFFromNexus.cpp:    localWorkspace->getInstrument()->getParameterMap()->clear();
Framework/SINQ/src/PoldiUtilities/PoldiSourceSpectrum.cpp:      getSpectrumParameter(source, poldiInstrument->getParameterMap());
```

martyngigg, 2017-02-15:
If we switch the return to `const ParameterMap &` or `shared_ptr<const ParameterMap>` would that avoid this problem?

SimonHeybrock, 2017-02-15:
Yes, but we would need to fix several algorithms. I tried providing a const *and* a non-const version and I get some compilation failures.

closed 2021-03-03 by stale[bot]