mantidproject/mantid #18862: Input workspaces can be forwarded to child algorithms for writing, without locking.

SimonHeybrock, 2017-02-15:
There does not seem to be a mechanism to prevent setting an input workspace as an inout workspaces or output workspace in a child algorithm. Apart from causing modifications of the input, the main problem is as follows:

- `Algorithm` locks input workspaces for reading. There can be multiple readers.
- Locking for child algorithms is skipped (it is assumed that correct locking has been done for the inputs/outputs of the parent, or workspaces are internal to the parent).

The consequence is a broken locking mechanism that can lead to crashed, or worse, unpredictable results of another algorithm being run at the same time.

Examples can be found in `DiffractionEventCalibrateDetectors`.

martyngigg, 2017-02-15:
I seem to remember a long time ago that there was talk, by Roman I think, that input properties should only be able to give access to `const` objects. The idea either got rejected for some reason or never followed through.

Would this be solved if we had such a concept? If inputs could not be modified then the parent doing the locking would be okay wouldn't it?

SimonHeybrock, 2017-02-15:
@martyngigg I think that is not enough since you can pass the workspace by-name to the child algorithm?

martyngigg, 2017-02-15:
As long as the name goes through the `WorkspaceProperty` wouldn' the `const`-only then apply to the child as well so that it couldn't modify it?

You are right with the name though that it could always be retrieved from the global `ADS`  :disappointed:

SimonHeybrock, 2017-02-15:
Right, that is exactly what is happening here: https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/DiffractionEventCalibrateDetectors.cpp#L143 :(

closed 2021-03-03 by stale[bot]