mantidproject/mantid #18987: Indirect ConvFit - truncated delta function

louisemccann, 2017-02-23:
When using ConvFit, there is no parameter for PeakRadius (this was implemented in 3.9 - see PR #18137)
This means the default value of 0 is used and gives a strange looking result:
[image]
The delta function (cyan) should be smooth but it is truncated.

SpencerHowells, 2017-02-23:
Default 0 should use the full x- range. How does a delta function work? Does it just scale the resolution? If so it would explain the shape because it has the x- range  of the res function.

louisemccann, 2017-02-24:
The truncated data isn't to do with the peak radius. I will make a separate issue for this

martyngigg, 2017-02-24:
@louisemccann As Spencer says the default of 0 should use the whole range. Where has the value of `50` come from?

SpencerHowells, 2017-02-24:
In the old version I would recommend setting PeakRadius=50 ( a large number about full energy range). I think it would be best to explicitly set a value rather than just assume default. So it could be 0 or 50. Perhaps 0 might be best.

louisemccann, 2017-02-24:
In that case this issue can be closed

closed 2017-02-24 by louisemccann